Next-Flip/Momentum-Firmware #22: SubGhz lost in menu

ldxjy, 2024-03-11:
### Describe the bug.

I flashed Momentum-Firmware from unleashed 071e directly.
Now SubGhz has disappared in menu( click center button in desktop).
I tried to reset in MNTM -> interface -> Mainmenu -> Reset menu. after Reboot. SubGhz has appared in menu.
But, after I edited mainmenu in MNTM -> interface -> Mainmenu， subghz has disappared again. In MNTM -> interface -> Mainmenu I saw subghz is in the App List.
[image]

### Reproduction

1. reset in MNTM -> interface -> Mainmenu -> Reset menu
2. reboot
3. edit MNTM -> interface -> Mainmenu app list
4. SubGhz disappared

Willy-JL, 2024-03-11:
Yes I noticed this myself too, it was an oversight when migrating app names. It is fixed in latest dev branch, you can use the dev builds from the website. Unfortunately, if you already modified the menu app list, you will need to remove subghz and rfid, and add them again. Only these 2 were affected, due to changing the app names back to match ofw

Willy-JL, 2024-03-11:
The issue was that the main menu migrated the app names, but not the momentum app when editing the config file, so only after editing with the app it broke

ldxjy, 2024-03-12:
> Yes I noticed this myself too, it was an oversight when migrating app names. It is fixed in latest dev branch, you can use the dev builds from the website. Unfortunately, if you already modified the [...]

ok, thanks, I will try dev firmware.

Willy-JL, 2024-05-12:
Fix was released in [mntm-002](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-002)

closed 2024-05-12 by Willy-JL