Next-Flip/Momentum-Firmware #23: stuck

Sarah831o, 2024-03-11:
### Describe the bug.

installing firmware and stuck on updating resources

### Reproduction

-

Willy-JL, 2024-03-11:
How long has it been stuck? It takes about 5 minutes overall to install if not more

Ghost6220, 2024-03-12:
Mine was stuck loading as well.  I tried installing today. I ended up disconnecting after 20 minutes

Willy-JL, 2024-03-12:
was it on flipper or on the website/qflipper that it got stuck?

UberGuidoZ, 2024-03-12:
It your SD card is loaded up with conflicting stuff, I’ve run into issues before. Try wiping the SD, installing Official, then upgrading to Momentum. Lastly, installing directly from the Flipper via a TGZ file can sometimes help.

Ghost6220, 2024-03-12:
I tried installing through web. I'll give the other way a shot.

Willy-JL, 2024-03-12:
as i asked, was it stuck on flipper as "updating resources", or was it stuck on web at "resources.tar"? they are different. one of them has an ongoing issue from OFW. please answer.

Willy-JL, 2024-03-19:
if it was stuck on flipper as "updating resources", then the problem is likely with the sd card. try rebooting, or formatting it.

if it stuck on web or qflipper at "resources.tar", then the problem is likely with USB CDC parameters. see #40.

closing as stale. feel free to comment with more info, but the above should likely cover most cases.

closed 2024-03-19 by Willy-JL