Next-Flip/Momentum-Firmware #25: Could you please make a video on how to use find my flipper for iphone? Thanks!

kenneth7398, 2024-03-11:
### Describe the bug.

Cant get find my iphone to work

### Reproduction

try find my iphone

Willy-JL, 2024-03-11:
You can find info in its folder at 
https://github.com/Next-Flip/Momentum-Firmware/tree/dev/applications/system/findmy

Or of course on the app repository 
https://github.com/MatthewKuKanich/FindMyFlipper

UberGuidoZ, 2024-03-12:
Also keep in mind this is a pretty new app and will likely be undergoing some changes and tweaks for a bit. If you’re unable to get it to work, make sure to provide as much info as possible so bugs can be worked out.

Willy-JL, 2024-03-30:
https://www.youtube.com/watch?v=ndfZD37tcs8
https://www.youtube.com/watch?v=XGwHmwvQoqo

closed 2024-03-30 by Willy-JL