Next-Flip/Momentum-Firmware #26: YardStick One emulation app

caquino, 2024-03-12:
### Description of the feature you're suggesting.

- An application for flipper that would allow the usage of [rfcat](https://github.com/atlas0fd00m/rfcat), making a "bridge" between the USB/Bluetooth communication to the CC1101

### Anything else?

I'm willing to try to implement it. Still, before starting, I would like someone with more knowledge about Flipper Zero's architecture and capabilities to let me know if this is even possible.

Willy-JL, 2024-03-12:
if i understand correctly, the objective is using the Flipper's CC1101 module remotely on a PC or other device, connected via USB or Bluetooth to the flipper. this program would have a communication layer to relay the commands to the flipper, plus of course a layer on the flipper side to accept and actuate these commands to the CC1101 module (possible either internal or external too).

i find this idea very interesting, and it sounds possible to me. if the program in question is already aware of how the cc1101 functions, it might just be a matter of writing the layer to translate calls from it to flipper api, over one of the 2 methods listed.

both USB and BT now support fully custom profile implementations as of mntm-001 and OFW 0.99-rc. this is used for example to implement mass storage, and the HID keyboard profiles via USB, and a serial profile and also HID keyboard over BT. the easiest solution i see for BT would be to use an existing serial profile, such as [the one in pc monitor](https://github.com/Next-Flip/Momentum-Apps/tree/dev/pc_monitor/helpers). for USB, much of the same too, tho i dont think there is a ready to use serial profile, the firmware uses its own one (as it does for BT) to communicate with qflipper and other companion apps, but it would need to be extracted and made into a standalone profile for it not to clash with the system one for companion apps (as was done recently for pc monitor).

also given this newfound full profile support, this could be portable as an application compatible with all firmwares! i think this could come out great. i dont think we would be able to get it in our TODOs any time soon, so please, go ahead and give it a shot! and ask here / on discord if you have any questions :D

caquino, 2024-03-14:
Your understanding is correct. The Yardstick One exposes a serial port that you use from RFCat to control the radio, and this opens up a lot of interesting exploration.

One example some years ago, I saw that the Wireless Village at Defcon had a CTF exercise where the teams had to shock each other captain using a dog training collar by reversing engineering the radio protocol and sending the signal, and I found it a really interesting and amusing idea and did some work using a Yardstick, you can see the code I wrote here: https://gist.github.com/caquino/270958014580ddbf297acce8ab9fe71f

So basically, the idea is what you described, similar to how we can use Flipper as a USB-serial adapter, which would be to expose via a serial port the protocol compatible with RFCat and control the CC1101 module using this interface.

Thanks for taking the time to give such a comprehensive answer to my question. I will try to allocate some of my free time to work on the app.
 
I have some homework to do as I also need to explore how RFCat works and document the protocol, but it should not be a problem.

Willy-JL, 2024-03-14:
Sounds great! I also realized after that serial is not the only way, with usb profiles you could make it directly emulate a yardstick, without the serial level of translating commands. An example is the mass storage app, which has a usb profile to emulate an SCSI device, definitely have a look there. I've also heard from the rest of the team that they're interested in this idea too, but we don't have much experience with usb descriptors and profiles either. But yeah if you can't find the time, let us know and we might eventually give it a shot outselves

caquino, 2024-03-14:
Thanks, @Willy-JL  I will jump on Discord after work and reach out to the team to see how we can collaborate. I would love to leverage the team's expertise as much as possible to avoid suffering for no reason.

I'm also reaching to the RFCat development team to see if this is something that interests them.

Crsarmv7l, 2025-01-15:
I proposed this to OFW long ago and they shot it down. The YS1 is amazing and my go to RF device even with its quirks. So much so that I have been developing my own, cheaper hardware with an mcu/cc1101/custom pcb and printed case that can be used either by phone or python.

Been stuck on a singular bug for the last few months though. Most signals sent over serial are accurate and tx fine....but 1 in 10 or so misses a byte or part of a byte with FIFO refills. I *think* it is related to the actual serial comms (found a bunch of fixes in espressif's board defs for serial issues but still need to test) .

Anyway all that to say, this would be amazing.