Next-Flip/Momentum-Firmware #31: Scripts (The js scripts)

Thesek2008, 2024-03-13:
### Description of the feature you're suggesting.

Can you add that you can switch between keyboard languages that i can use it because i am in germany and i cant realy use it effectifly

Willy-JL, 2024-03-13:
We plan to add the possibility to set a keyboard layout, check the projects tab.

Willy-JL, 2024-03-19:
as of #33 there is now `badusb.altPrint()` and `badusb.altPrintln()`, which allow layout-independent functionality __only on windows__. still plan to implement layouts, as that would work on macos, linux, android and ios too.

Willy-JL, 2024-05-12:
Was released in [mntm-002](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-002)

closed 2024-05-12 by Willy-JL