Next-Flip/Momentum-Firmware #32: Native JavaScript

Pro2stinger, 2024-03-13:
### Describe the enhancement you're suggesting.

Could you add more native javascript features such as setTimeout, setInterval, Promise, etc?

### Anything else?

All variables are of static type and I cant use `==` instead I'm forced to do `===`

Willy-JL, 2024-03-13:
The short version is no.

This is mjs, not js. Mjs doesn't support these, and adding support for them would not be possible. This is intended only to give some scripting, its not a full js implementation nor does it want to be. To support these, it would require too much overhead and dfu flash space, all of which we do not have on a 64mhz cpu, 256kb ram, and 1mb flash that is already full.

closed 2024-03-13 by Willy-JL

Willy-JL, 2024-03-13:
To add more context, all of the features you listed would require an implementation of an event loop, timer tasks, threading (possibly) and async. This is simply beyond the scope of what is possible, and intended to be possible.

Value semi-coalescence (==) is also a sin and superfluous, and adds unnecessary complexity and space used to support something that has no place existing anyway.