Next-Flip/Momentum-Firmware #34: It is not possible to read the Desfire part of a badge equipped with a 125 kHz and 13.56 MHz chip.

Katsugami, 2024-03-14:
### Describe the bug.

Hello,

I have a badge that integrates a Mifare Desfire chip at 13.56 MHz as well as an EM4100 [EM-Micro] chip in a single badge.

For reading the 125 kHz part, there's no issue, the Flipper reads it immediately. However, when trying to read the Desfire part, the Flipper remains stuck in continuous reading mode with the message "Don't Move".

I believe that the 13.56 MHz antenna of the Flipper might be disrupted because it is simultaneously powering both chips. The response of the badge at the 125 kHz frequency could be causing interference.

Do you think this can be corrected?

### Reproduction

1. Having a badge with a MIFARE Desfire chip and a 125 kHz chip.
2. Using the Flipper's NFC application for reading.
3. Pressing "Read".
4. The application remains stuck on "Don't Move".

Another scenario:
3. Performing an additional action.
4. Selecting "Read Specific Card Type".
5. Choosing "Read MIFARE DESFire".
6. The application remains stuck on "Don't Move".

### Target

NFC and RFID

Willy-JL, 2024-03-14:
interesting issue... you might be right in your guess, can't think of anything else myself. desfire isnt cracked anyway, so emulating would not work, but still it is weird that it remains stuck.

does this happen on official firmware too? please try on 0.99-rc

Katsugami, 2024-03-14:
I have just tested with the latest official firmware 99.1, and the issue persists.

Here is a demonstration video: https://youtu.be/iGWESNVlHbA

Willy-JL, 2024-04-05:
@Katsugami can you check on latest dev version?

Katsugami, 2024-04-05:
How can I install this version?

Willy-JL, 2024-04-05:
https://momentum-fw.dev/update

Katsugami, 2024-04-05:
I just tested it and it's still the same.

jensMF, 2024-05-03:
I have a tag that was recognized as Mifare Desfire when I used the NFC read operation. After my last update, it is recognized as an iso14something and when I try to read it as Mifare Desfire the flipper does not recognize it.
May be this helps to find the issue?

mirceaagr, 2024-05-10:
I have the same issue with Unleashed. I don't think it's linked to Momentum

zinongli, 2024-05-25:
I am having similar issue. One transit card that I own can only be read as 4A Unknown and having SAK of 20, so it has to be DESfire or Plus. I found another person's reading of this type of card online done with NXP Taginfo confirming it's DESfire. However my flipper won't read it.

zinongli, 2024-05-25:
This problem seems to have been resolved in OFW. I flashed my firmware to OFW 0.101.2 and have no issue reading the aforementioned DESfire card. It seems to be an older issue since NFC refactoring.

Willy-JL, 2024-05-28:
> I am having similar issue. One transit card that I own can only be read as 4A Unknown and having SAK of 20, so it has to be DESfire or Plus. I found another person's reading of this type of card online done with NXP Taginfo confirming it's DESfire. However my flipper won't read it.

> This problem seems to have been resolved in OFW. I flashed my firmware to OFW 0.101.2 and have no issue reading the aforementioned DESfire card. It seems to be an older issue since NFC refactoring.

to bring closure here too, we discussed the issue on discord and got a fix for desfire not reading at all with 4931b5b2f7e07502c15759d0934b494ff8830011, so @zinongli @mirceaagr @jensMF your issue should be resolved now, please try latest dev build at https://momentum-fw.dev/update

however, the original issue reported by @Katsugami doesnt seem to be related. what the other people in this thread reported is a regression in 75ece9b697d1e3b66a75f2d423e51f369bc387f1 that i fixed in 4931b5b2f7e07502c15759d0934b494ff8830011. but @Katsugami appears to have never been able to read his tag. please, try again on latest dev build too just to make sure, if you are still not able to read it then this is likely something you should ask upstream at https://github.com/flipperdevices/flipperzero-firmware

Katsugami, 2024-06-22:
Hello,

I have just updated to the latest version of mumuntum (Release).

I have tested the badge in question, and it works perfectly. It can now read both 125K and 13.56.

I believe we can close this issue.

Thank you again.

closed 2024-06-22 by Willy-JL