Next-Flip/Momentum-Firmware #36: Comparison to XFW

lordrak007, 2024-03-15:
### Description of the feature you're suggesting.

Hi Willy,
as you are (or was?) big supporter on XFW, why you make your own fork?
Which beneficts has your MFW compared to XFW? You have compared Unsleashed and original and so on, but mission this comparisson and reason.
Thank you
Lordrak

Willy-JL, 2024-03-15:
i was the main developer on XFW along with Sil, together we made the features that made XFW "XFW". the project owner was only "the owner" and neglected the project, finally deciding remove me over bogus reasons, also bringing personal issues into business and slandering me publicly over made-up statements and leaking framed DMs.

therefore, i moved on with the same codebase that i have been developing for over 1 year, and rebranded. you can see for yourself that currently there isn't much going on in XFW anymore except updating from unleashed. i truly wish XFW can get back on its feet and be a worthy competitor, but as it stands now, the spirit that drove XFW's innovation is now in Momentum.

since the split a few weeks ago, there's been no real development tied to the XFW name yet. over on Momentum, we have already made many new JavaScript API modules, new VGM color options, the FindMy Flipper app (Matthew K is part of our team and we worked together on it, though app ownership remains under his name), along with many improvements and fixes.

essentially, if you liked XFW for the features and improvements and stability, the lot who made that possible are now over here under Momentum. remains to be seen how XFW will continue.

lordrak007, 2024-03-15:
Ok, thanks you for explanation very much.
Crossed fingers.
Lordrak

closed 2024-03-15 by lordrak007

bmyhny, 2024-03-17:
Thank you I'll now switch over as at first I thought it was just a ripoff but I understand now! Awesome firmware looking forward to future of MFW!