Next-Flip/Momentum-Firmware #37: git clone won't work

Nothing2Do, 2024-03-16:
### Describe the bug.

I have this on my shell : "{10:36}~/flipper ➭ git clone --recursive --jobs 8 https://github.com/Next-Flip/Momentum-Firmware.git
Cloning into 'Momentum-Firmware'...
remote: Enumerating objects: 146733, done.
remote: Counting objects: 100% (2075/2075), done.
remote: Compressing objects: 100% (817/817), done.
error: 677 bytes of body are still expected67 MiB | 564.00 KiB/s
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
" or some other error ("error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)", "error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.", etc ...).
So, I can't clone your repositori ...

### Reproduction

git clone --recursive --jobs 8 https://github.com/Next-Flip/Momentum-Firmware.git

Willy-JL, 2024-03-16:
We can't do anything about that, sounds like something it's going wrong with your network stack, either packet loss or general instability. Try on another network or another devices maybe

Nothing2Do, 2024-03-16:
I know you'll say that, I'll come back later !
So, problem on my end, but only on with your git(hub) account ...

closed 2024-03-16 by Nothing2Do