Next-Flip/Momentum-Firmware #40: Error on firmware install: resources.tar operation timeout

spaceblankey, 2024-03-16:
### Describe the bug.

[image]

On a fresh 32gb SD card with stock flipper zero firmware, latest release (downloaded today), on Windows 11

### Reproduction

Install using Qflipper on windows 11, select Zip file downloaded from Github

### Logs

```Text
[APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T19:40:31
[APP] OS info: Windows 11 Version 22H2 11 10.0.22621 Qt 6.4.2
[RPC] Starting RPC session...
[RPC] Failed to start RPC session: Failed to open serial port: Access is denied.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) System Protobuf Version START
[RPC] (1) System Protobuf Version SUCCESS
[RPC] (2) Property Get START
[RPC] (2) Property Get SUCCESS
[RPC] (3) Storage Info @/ext START
[RPC] (3) Storage Info @/ext SUCCESS
[RPC] (4) Storage Stat @/ext/Manifest START
[RPC] (4) Storage Stat @/ext/Manifest SUCCESS
[RPC] (5) System Get DateTime START
[RPC] (5) System Get DateTime SUCCESS
[RPC] (6) System Set DateTime START
[RPC] (6) System Set DateTime SUCCESS
[RPC] Stopping RPC session...
[RPC] RPC session stopped successfully.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) Gui Start ScreenStream START
[RPC] (1) Gui Start ScreenStream SUCCESS
[RPC] (2) Gui Start VirtualDisplay START
[RPC] (2) Gui Start VirtualDisplay SUCCESS
[DEV] Full Update @Algl4t START
[UTL] Region Provisioning @Algl4t START
[UTL] Region Provisioning @Algl4t SUCCESS
[UTL] Refresh Storage Info @Algl4t START
[RPC] (3) Storage Info @/ext START
[RPC] (3) Storage Info @/ext SUCCESS
[RPC] (4) Storage Stat @/ext/Manifest START
[RPC] (4) Storage Stat @/ext/Manifest SUCCESS
[UTL] Refresh Storage Info @Algl4t SUCCESS
[UTL] Create Path @/ext/update/f7-update-dev-11032024-022fccf0 START
[RPC] (5) Storage Stat @/ext/update START
[RPC] (5) Storage Stat @/ext/update SUCCESS
[RPC] (6) Storage MkDir @/ext/update START
[RPC] (6) Storage MkDir @/ext/update SUCCESS
[RPC] (7) Storage MkDir @/ext/update/f7-update-dev-11032024-022fccf0 START
[RPC] (7) Storage MkDir @/ext/update/f7-update-dev-11032024-022fccf0 SUCCESS
[UTL] Create Path @/ext/update/f7-update-dev-11032024-022fccf0 SUCCESS
[UTL] Upload 6 entries @Algl4t START
[RPC] (8) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/firmware.dfu START
[RPC] (8) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/firmware.dfu SUCCESS
[RPC] (9) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/radio.bin START
[RPC] (9) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/radio.bin SUCCESS
[RPC] (10) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/resources.tar START
[RPC] (10) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/resources.tar SUCCESS
[RPC] (11) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/splash.bin START
[RPC] (11) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/splash.bin SUCCESS
[RPC] (12) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/update.fuf START
[RPC] (12) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/update.fuf SUCCESS
[RPC] (13) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/updater.bin START
[RPC] (13) Storage Write @/ext/update/f7-update-dev-11032024-022fccf0/updater.bin SUCCESS
[UTL] Upload 6 entries @Algl4t SUCCESS
[UTL] Start Update @Algl4t START
[RPC] (14) System Update @/ext/update/f7-update-dev-11032024-022fccf0/update.fuf START
[RPC] (14) System Update @/ext/update/f7-update-dev-11032024-022fccf0/update.fuf SUCCESS
[RPC] (15) System Reboot START
[RPC] (15) System Reboot SUCCESS
[RPC] Serial connection was lost.
[RPC] Stopping RPC session...
[RPC] RPC session stopped successfully.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) System Protobuf Version START
[RPC] (1) System Protobuf Version SUCCESS
[RPC] (2) Property Get START
[RPC] (2) Property Get SUCCESS
[RPC] (3) Storage Info @/ext START
[RPC] (3) Storage Info @/ext SUCCESS
[RPC] (4) Storage Stat @/ext/Manifest START
[RPC] (4) Storage Stat @/ext/Manifest SUCCESS
[RPC] (5) System Get DateTime START
[RPC] (5) System Get DateTime SUCCESS
[RPC] (6) System Set DateTime START
[RPC] (6) System Set DateTime SUCCESS
[RPC] Stopping RPC session...
[RPC] RPC session stopped successfully.
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[UTL] Start Update @Algl4t SUCCESS
[DEV] Full Update @Algl4t SUCCESS
[RPC] (16) Gui Start ScreenStream START
[RPC] (16) Gui Start ScreenStream SUCCESS
[RPC] (17) Storage Info @/ext START
[RPC] (17) Storage Info @/ext SUCCESS
[DEV] Full Update @Algl4t START
[RPC] (18) Gui Start VirtualDisplay START
[UTL] Region Provisioning @Algl4t START
[RPC] (18) Gui Start VirtualDisplay SUCCESS
[RPC] (19) Storage Write @/int/.region_data START
[RPC] (19) Storage Write @/int/.region_data SUCCESS
[UTL] Region Provisioning @Algl4t SUCCESS
[UTL] Refresh Storage Info @Algl4t START
[RPC] (20) Storage Info @/ext START
[RPC] (20) Storage Info @/ext SUCCESS
[RPC] (21) Storage Stat @/ext/Manifest START
[RPC] (21) Storage Stat @/ext/Manifest SUCCESS
[UTL] Refresh Storage Info @Algl4t SUCCESS
[UTL] Create Path @/ext/update/f7-update-mntm-001 START
[RPC] (22) Storage Stat @/ext/update START
[RPC] (22) Storage Stat @/ext/update SUCCESS
[RPC] (23) Storage Stat @/ext/update/f7-update-mntm-001 START
[RPC] (23) Storage Stat @/ext/update/f7-update-mntm-001 SUCCESS
[RPC] (24) Storage MkDir @/ext/update/f7-update-mntm-001 START
[RPC] (24) Storage MkDir @/ext/update/f7-update-mntm-001 SUCCESS
[UTL] Create Path @/ext/update/f7-update-mntm-001 SUCCESS
[UTL] Upload 7 entries @Algl4t START
[RPC] (25) Storage Write @/ext/update/f7-update-mntm-001/firmware.dfu START
[RPC] (25) Storage Write @/ext/update/f7-update-mntm-001/firmware.dfu SUCCESS
[RPC] (26) Storage Write @/ext/update/f7-update-mntm-001/firstboot.bin START
[RPC] (26) Storage Write @/ext/update/f7-update-mntm-001/firstboot.bin SUCCESS
[RPC] (27) Storage Write @/ext/update/f7-update-mntm-001/radio.bin START
[RPC] (27) Storage Write @/ext/update/f7-update-mntm-001/radio.bin SUCCESS
[RPC] (28) Storage Write @/ext/update/f7-update-mntm-001/resources.tar START
[UTL] Upload 7 entries @Algl4t ERROR: Operation timeout (generic)
[DEV] Full Update @Algl4t ERROR: Operation timeout (generic)
[RPC] (28) Storage Write @/ext/update/f7-update-mntm-001/resources.tar ERROR: Operation timeout (generic)
[RPC] (32) Gui Stop VirtualDisplay START
[DEV] Full Update @Algl4t START
[UTL] Region Provisioning @Algl4t START
[RPC] (32) Gui Stop VirtualDisplay ERROR: Operation timeout (generic)
```

Willy-JL, 2024-03-16:
yup, problem on OFW. if you want to make yourself heard on the matter, here https://github.com/flipperdevices/flipperzero-firmware/issues/3452

Willy-JL, 2024-03-16:
in the meantime, try downgrading to ofw 0.97.1 ([tgz download](https://update.flipperzero.one/builds/firmware/0.97.1/flipper-z-f7-update-0.97.1.tgz)), then installing momentum again

feldlef, 2024-03-20:
Samething.
Repaired Flipper using qflipper
ran Momentum file and got error.

`326 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T10:40:31
326 [APP] OS info: Fedora Linux 39 (Workstation Edition) 39 6.7.9-200.fc39.x86_64 Qt 6.4.3
425 [REG] Detected new device: VID_0x483:PID_0xdf11
464 [DBG] Device reported transfer size: 1024
464 [DBG] Bytes uploaded: 32 100%
464 [DBG] Upload has finished.
464 [REG] Registering the device
466 [BKD] Current device changed to "Lagplav"
674 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
937 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
3749 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
3800 [DEV] Full Repair @Lagplav START
4725 [ZIP] Uncompressing file with size of 168264 bytes...
4726 [ZIP] Uncompression finished : No Error
5283 [ZIP] Uncompressing file with size of 351765 bytes...
5294 [ZIP] Uncompression finished : No Error
5960 [RCY] Set Recovery boot mode @Lagplav START
6057 [DBG] Device is not idle, resetting...
6082 [DBG] Device reported transfer size: 1024
6083 [DBG] Bytes uploaded: 128 100%
6083 [DBG] Upload has finished.
6083 [DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
6083 [DBG] Device is not idle, resetting...
6108 [DBG] Device reported transfer size: 1024
7384 [DBG] Failed to perform control transfer: LIBUSB_ERROR_NO_DEVICE
7384 [DBG] Unable to get device status
7384 [DBG] An error has occurred during download phase
7384 [DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
7384 [DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
7384 [DBG] Failed to release interface: LIBUSB_ERROR_NO_DEVICE
7384 [DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
7384 [REG] Device went offline: VID_0x483:PID_0xdf11
7387 [REG] Detected new device: VID_0x483:PID_0xdf11
7421 [DBG] Device reported transfer size: 1024
7421 [DBG] Bytes uploaded: 32 100%
7421 [DBG] Upload has finished.
7421 [REG] Device went back online
7483 [RCY] Set Recovery boot mode @Lagplav SUCCESS
7483 [RCY] Co-Processor Firmware Download @Lagplav START
7484 [DBG] Device is not idle, resetting...
7509 [DBG] Device reported transfer size: 1024
7510 [DBG] Bytes uploaded: 2 100%
7510 [DBG] Upload has finished.
7510 [DBG] FUS appears NOT to be running: Error occurred, Not running.
7510 [DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
7510 [DBG] Device is not idle, resetting...
7536 [DBG] Device reported transfer size: 1024
8811 [DBG] Failed to perform control transfer: LIBUSB_ERROR_NO_DEVICE
8811 [DBG] Bytes uploaded: 0 0%
8811 [DBG] End of transmission.
8812 [DBG] Upload has finished.
8812 [DBG] Failed to read FUS status
8812 [DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
8812 [DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
8812 [DBG] Failed to release interface: LIBUSB_ERROR_NO_DEVICE
8812 [DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
8812 [REG] Device went offline: VID_0x483:PID_0xdf11
8818 [REG] Detected new device: VID_0x483:PID_0xdf11
8844 [DBG] Device reported transfer size: 1024
8844 [DBG] Bytes uploaded: 32 100%
8844 [DBG] Upload has finished.
8844 [REG] Device went back online
8903 [DBG] Device is not idle, resetting...
9731 [REG] Device went offline: VID_0x483:PID_0xdf11
10131 [DBG] Getting Co-Processor (Wireless) status...
10131 [DBG] Failed to get FUS status. The device is offline at the moment.
[... 2093 lines cut ...]
122264 [RPC] (19) Storage Md5Sum @/ext/update/f7-update-mntm-001/firstboot.bin START
122326 [DBG] File is identical: "/ext/update/f7-update-mntm-001/firstboot.bin"
122326 [RPC] (19) Storage Md5Sum @/ext/update/f7-update-mntm-001/firstboot.bin SUCCESS
122326 [RPC] (20) Storage Md5Sum @/ext/update/f7-update-mntm-001/radio.bin START
122618 [DBG] File is identical: "/ext/update/f7-update-mntm-001/radio.bin"
122618 [RPC] (20) Storage Md5Sum @/ext/update/f7-update-mntm-001/radio.bin SUCCESS
122618 [RPC] (21) Storage Md5Sum @/ext/update/f7-update-mntm-001/resources.tar START
122702 [DBG] File changed: "/ext/update/f7-update-mntm-001/resources.tar" old: "d41d8cd98f00b204e9800998ecf8427e" new: "6d70f06433ce7ee6ccc51ce4d399ef07"
122702 [RPC] (21) Storage Md5Sum @/ext/update/f7-update-mntm-001/resources.tar SUCCESS
122702 [RPC] (22) Storage Md5Sum @/ext/update/f7-update-mntm-001/splash.bin START
122754 [DBG] File does not exist: "/ext/update/f7-update-mntm-001/splash.bin"
122754 [RPC] (22) Storage Md5Sum @/ext/update/f7-update-mntm-001/splash.bin SUCCESS
122754 [RPC] (23) Storage Md5Sum @/ext/update/f7-update-mntm-001/update.fuf START
122807 [DBG] File does not exist: "/ext/update/f7-update-mntm-001/update.fuf"
122807 [RPC] (23) Storage Md5Sum @/ext/update/f7-update-mntm-001/update.fuf SUCCESS
122807 [RPC] (24) Storage Md5Sum @/ext/update/f7-update-mntm-001/updater.bin START
122860 [DBG] File does not exist: "/ext/update/f7-update-mntm-001/updater.bin"
122860 [RPC] (24) Storage Md5Sum @/ext/update/f7-update-mntm-001/updater.bin SUCCESS
122860 [UTL] Verify checksum of 7 items @Lagplav SUCCESS
122860 [UTL] Upload 4 entries @Lagplav START
122861 [RPC] (25) Storage Write @/ext/update/f7-update-mntm-001/resources.tar START
153132 [UTL] Upload 4 entries @Lagplav ERROR: Operation timeout (generic)
153132 [DEV] Full Update @Lagplav ERROR: Operation timeout (generic)
153132 [BKD] Current operation finished with error: "Operation timeout (generic)"
153133 [RPC] (25) Storage Write @/ext/update/f7-update-mntm-001/resources.tar ERROR: Operation timeout (generic)`

Willy-JL, 2024-03-20:
@feldlef read what i said to the other person, right above your comment. follow those instructions.

dalaeance, 2024-03-27:
Not sure if it was just a fluke, or what, but I was having this issue. I tried twice, with failures. Before my third attempt, I increased my screen lock timeout from 2 minutes to 10 minutes, on a hunch, and it worked. I didn't test it scientifically, so it may have been a coincidence. Figured I'd mention it anyway.

syntaxHax, 2024-04-09:
had the same error

resolution:
- formatted the sd card (fat32)

next attempt went smoothly

Willy-JL, 2024-06-13:
Update: OFW has now rectified the issue, and all CFWs either have the new proper fix to this issue (Momentum, Unleashed, probably RM soon enough) or the old USB CDC revert patch (XFW). OFW is still affected by this bug in 0.102.3, but is fixed in dev branch and will be fixed from next OFW release.

If you are affected by this issue: select "development" channel in qflipper / phone app, and update to latest OFW dev; then, update to Momentum / whatever FW you want.

Willy-JL, 2024-07-10:
Fix released in OFW 0.103.1, if necessary update to OFW if you have issues installing Momentum or any other CFW. After that, CFW updates should be fine and not affected by this bug.

closed 2024-07-10 by Willy-JL