Next-Flip/Momentum-Firmware #42: SubGhz: Add autosave for new frequencies

Zarcolio, 2024-03-18:
### Describe the enhancement you're suggesting.

I have a doorbell that changes the frequency every time it rings. It would be nice if the Frequency Analyzer could automatically save an analysis of captured signals of freshly captured frequencies. 

### Anything else?

Momentum really rocks 😁

Sil333033, 2024-03-18:
what frequencies can you see? Your doorbell could be using an [FM]https://en.wikipedia.org/wiki/Frequency_modulation#/media/File:Waterfall_FM.jpg)(Frequency modulation)

Zarcolio, 2024-03-18:
> what frequencies can you see? Your doorbell could be using an [FM](https://en.wikipedia.org/wiki/Frequency_modulation#/media/File:Waterfall_FM.jpg) (Frequency modulation)

My F0 shows AM for these signals, so no FM.
I managed to read 2 signals at frequencies 433.88 and 434.07, both AM

Willy-JL, 2024-03-19:
frequency analyzer is not ideal for this kind of objective. youre looking for hopping mode in read menu instead. effectively, hopping mode does the same as frequency analyzer, but stops for longer on frequencies where RSSI is high to try and capture a signal.

closed 2024-03-24 by Zarcolio