Next-Flip/Momentum-Firmware #43: SubGhz: ignore frequencies temporarily /permanently

Zarcolio, 2024-03-18:
### Describe the enhancement you're suggesting.

In my neighborhood a couple of frequencies are captured every couple of seconds. Of course it's possible to use RSSI to tweak what strength of signals to capture. But it would be so much easier to just temporarily or (semi) permanently ignore a frequency. 

### Anything else?

As I said before, love this fw 😁

Sil333033, 2024-03-18:
How did you determine this behaviour? with an proper SDR or the Frequency Analyzer on the Flipper?

Zarcolio, 2024-03-18:
> How did you determine this behaviour? with an proper SDR or the Frequency Analyzer on the Flipper?

I constantly see 2 frequencies in the Frequency Analyze adding up the the number of times it has been captured.

Willy-JL, 2024-04-05:
the idea needs some more fleshign out as to how it could be implemented UI wise. but it would be a nice addition, yes

Willy-JL, 2024-05-18:
Now that I think about it, you might be able to do this via Momentum > Protocols > SubGhz freqs.

Frequency analyzer uses the same list of frequencies as read does, as far as I know. So you could set Use Defaults to off, then add the frequencies you want to check inside the Static Freqs > Add static freq

X3r0C00L, 2024-05-31:
The Default frequency values are stored in the subghz_setting.c file. 
I havent traversed the rest of the files in the build to see where the original call is made to this c file to get the defaults. But I assume that when using the frequency analyzer it simply runs maybe a while loop or some type of loop and loops through the defaults list so many times per second until something kicks it out of the loop such as an increase in RSSI on a certain frequency. Since this app can also read from the Static frequencies that are manually added, couldnt an almost identical menu item be created in protocols as the static frequencies that would allow a user to input frequencies to skip. 
Then just add some if/then to the original loop in the frequency analyzer to check each scanned frequency to see if it is in the skip list/array and ignore it. 
Please dont bash me, I have only looked at the single file as of this moment and  have not looked at the rest of the code. So at this point I am only speculating on current function.