Next-Flip/Momentum-Firmware #44: infrared functions didn't work when flipper pluging a sub-ghz module

ldxjy, 2024-03-19:
### Describe the bug.

sub-ghz module I use :https://www.youtube.com/watch?v=WM2bWjn5jOY&list=PLwfLF8jiGePKG0yLy8_yOxssiAvejQtoO&index=1&t=2203s&pp=gAQBiAQB
firmware version:mntm-dev-6ae098fa

unleashed 071e works like normal.

### Reproduction

1、pluging sub-ghz module on flipper
2、run infrared app - universal remote -TVs - Power button
3、IR remote won't work normally,EXT sub-ghz module LED blinking

### Target

bug fix

Willy-JL, 2024-03-19:
you have autodetect for ir blaster modules on, which is by default. unfortunately the only way to detect ir blasters also detects subghz modules due to how they work at a basic hardware level. unplug the module, or disable autodetect

ldxjy, 2024-03-19:
> you have autodetect for ir blaster modules on, which is by default. unfortunately the only way to detect ir blasters also detects subghz modules due to how they work at a basic hardware level. unplug the module, or disable autodetect

try unleashed firmware 071e，it works normal.
infrared functions works well when flipper pluging a sub-ghz module

Willy-JL, 2024-03-19:
i will refer you to my previous answer, which explains why that is.

> you have autodetect for ir blaster modules on, which is by default. unfortunately the only way to detect ir blasters also detects subghz modules due to how they work at a basic hardware level. unplug the module, or disable autodetect

please, read when someone answers you.

ldxjy, 2024-03-19:
> i will refer you to my previous answer, which explains why that is.
>
>
> please, read when someone answers you.

I did read your answer, Im living in non-english country.We may unserstanding things in different ways.
I think flipper should works like unleashed firmware 071e. 
Momentum-Firmware infrared functions didn't work when flipper pluging a sub-ghz module, but unleashed firmware 071e works well, that made me think it is a bug. It just software things, If it is hardware things, all custom firmware will work same.

Willy-JL, 2024-03-19:
my answer is very simple. the key part is "you have autodetect for ir blaster modules on". it means that you have the "autodetect" feature enabled. this "autodetect", and the name implies, will automatically detect any infrared blasters. subghz modules are not the only modules. there are infared modules. on momentum, the infrared modules are automatically detected. as i said, "which is by default". this automatic detection is enabled by default. as i said, you can "unplug the module", or you can "disable autodetect". autotomatic detection is possible to disable. do that.

ldxjy, 2024-03-19:
> > you have autodetect for ir blaster modules on, which is by default. unfortunately the only way to detect ir blasters also detects subghz modules due to how they work at a basic hardware level. unplug the module, or disable autodetect
> 
> try unleashed firmware 071e，infrared functions works well when flipper pluging a sub-ghz module. Reproduction 1、pluging sub-ghz module on flipper 2、run infrared app - universal remote -TVs - Power button 3、IR remote works normally,EXT sub-ghz module LED stay off.
> 
> unleashed firmware 071e autodetect is off by default?**where I can find this setting?** but if autodetect is off by default, sub-ghz module won't work when I use Sub-Ghz app right?

I do unserstand your answer clearly.
How to disable autodetect on momentum？
I did ask in previous reply.

Willy-JL, 2024-03-19:
infrared > gpio options > autodetect off

ldxjy, 2024-03-19:
well, Im flashing flipper back to momentum.I think I should find this setting by myself.But it flashing slowly.
you reply fast like lightning.haha~
It has different settings from unleashed firmware!I should check the infrared app settings frist.
thanks!

closed 2024-03-19 by Willy-JL