Next-Flip/Momentum-Firmware #46: Update of Apps not working

Zaggan, 2024-03-19:
### Describe the bug.

Whenever an app already installed gets and update, it allows to press the update button, but does not install it.

### Reproduction

1. Open official Flipper Zero app in Android, latest release (Samsung S23 Ultra)[not sure if same happens in IOS].
2. Click on Hub, and a green notification shows Updates are available.
3. Click on Installed.
4. Either click update all or update individually the available apps
5. Progress bar reaches 100%
6. App state returs to update button to "update available" and update is not installed

### Target

Update apps already installed

### Anything else?

This is tested specifically with Video Game Module Tool

Willy-JL, 2024-03-19:
yup, thats because youre trying to use the flipper app to update something that is installed with the firmware. as a general rule of thumb, apps that come with the firmware should be updated with the firmware. ones included with the firmware are built to integrate with the firmware better, and in some cases have extra features or integrations too. apps are all at latest version when we make an update, if you want use the dev builds of momentum to get even newer versions, but just because the flipper app says there is an update available doesnt mean there actually is.

closed 2024-03-19 by Willy-JL

Zaggan, 2024-03-20:
I appreciate the time for this explanation.