Next-Flip/Momentum-Firmware #48: Stuck on lockscreen after down-grading to stock firmware and then re-updating to momentum. Cannot enter screen with prompt to enter pin.

aoxhwjfoavdlhsvfpzha, 2024-03-20:
### Describe the bug.

After re-upgrading to momentum I get locked out of the flipper, nothing I press does anything, and I cannot get the prompt to enter my pin. 

### Reproduction

1. Have a pin set in momentum
2. Go back to stock firmware
3. Go back to momentum
4. Attempt to use the flipper

### Anything else?

I had the pin set in Xtreme before updating to momentum, can't say for certain this problem exists if you first set the pin within momentum.

I have tried downgrading a second time, setting a pin in the default firmware, and then re-updating to momentum a second time, and still had the same issue.

I have also tried going back to stock, disabling that pin, and then going back to momentum again with no luck.

aoxhwjfoavdlhsvfpzha, 2024-03-20:
Just

closed 2024-03-20 by aoxhwjfoavdlhsvfpzha

Willy-JL, 2024-03-20:
it does default to those options. you probably had them already disabled (you can turn off the prompt to unlock) on xtreme, which carried over to momentum.