Next-Flip/Momentum-Firmware #49: created Mass Storage not appearing on Slackware 15.0 Linux

leafy-psx, 2024-03-22:
### Describe the bug.

Created disk image 32M test.img yesterday, and it popped up in KDE Plasma and asked to mount, but today when I select it, no pop-up in KDE. I tried creating 32M qwerty.img, then selecting, but nothing. I checked dmesg as root and I don't see anything there.

### Reproduction

1. Open Mass Storage
2. Select Disk Image
3. scroll to test.img
4. hit OK button
5. screen shows USB Mass Storage (test)
6. unplug and re-plug USB cable

### Target

x86_64 Intel with Slackware 15.0

### Logs

```Text
13167316 [REG] Detected new device: VID_0x483:PID_0x5740
13167357 [DBG] Trying serial port flip_Crasile at /dev/ttyACM0
13167357 [DBG] Using  serial port flip_Crasile at /dev/ttyACM0
13167357 [RPC] Starting RPC session...
13167530 [RPC] RPC session started successfully.
13167530 [RPC] (1) System Protobuf Version START
13167651 [DBG] Detected protobuf version: 0.22
13167651 [RPC] (1) System Protobuf Version SUCCESS
13167651 [RPC] (2) Property Get START
13167963 [RPC] (2) Property Get SUCCESS
13167964 [RPC] (3) Storage Info @/ext START
13168084 [RPC] (3) Storage Info @/ext SUCCESS
13168086 [RPC] (4) Storage Stat @/ext/Manifest START
13168197 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
13168205 [RPC] (5) System Get DateTime START
13168313 [DBG] Flipper time skew is -902 milliseconds
13168314 [RPC] (5) System Get DateTime SUCCESS
13168332 [RPC] (6) System Set DateTime START
13168451 [RPC] (6) System Set DateTime SUCCESS
13168452 [RPC] Stopping RPC session...
13168490 [RPC] RPC session stopped successfully.
13168490 [REG] Registering the device
13168490 [DEV] Version: mntm-001 commit: 58bd6383 radio: 1.17.3
13168490 [BKD] Current device changed to ""
13168606 [RPC] Starting RPC session...
13168780 [RPC] RPC session started successfully.
13168781 [RPC] (1) Gui Start ScreenStream START
13168902 [RPC] (1) Gui Start ScreenStream SUCCESS
```

### Anything else?

not sure why it worked once yesterday :)

leafy-psx, 2024-03-22:
qFlipper was open because I was doing a lot of editing to a bad_kb ducky script and dragging it over.

That's probably my issue, I will test it later and report back.

leafy-psx, 2024-03-23:
OK, it mounts when qFlipper is closed. Is this a well known thing?

Unfortunately as soon as I exit the screen with the USB stick image on the Flipper, the virtual disk unmounts.

Is there not a way to keep it mounted then run bad_kb?

leafy-psx, 2024-03-23:
When you "Select Disk Image"
```
[  844.062041] usb 1-1.4: USB disconnect, device number 9
[  844.649109] usb 1-1.4: new full-speed USB device number 10 using ehci-pci
[  844.730467] usb 1-1.4: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 1.00
[  844.730483] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  844.730486] usb 1-1.4: Product: Crasile
[  844.730488] usb 1-1.4: Manufacturer: Flipper Devices Inc.
[  844.730490] usb 1-1.4: SerialNumber: qwerty
[  844.730916] usb-storage 1-1.4:1.0: USB Mass Storage device detected
[  844.731519] scsi host4: usb-storage 1-1.4:1.0
[  845.785861] scsi 4:0:0:0: Direct-Access     Flipper  Mass Storage     0001 PQ: 0 ANSI: 4
[  845.787345] sd 4:0:0:0: [sdc] 65536 512-byte logical blocks: (33.6 MB/32.0 MiB)
[  845.788079] sd 4:0:0:0: [sdc] Write Protect is off
[  845.788083] sd 4:0:0:0: [sdc] Mode Sense: 03 00 00 00
[  845.788711] sd 4:0:0:0: [sdc] No Caching mode page found
[  845.788714] sd 4:0:0:0: [sdc] Assuming drive cache: write through
[  845.807326]  sdc: sdc1
[  845.811499] sd 4:0:0:0: [sdc] Attached SCSI removable disk
```
Then I can mount /dev/sdc1, however if I hit back on the Flipper
```
[ 1003.511889] usb 1-1.4: USB disconnect, device number 10
[ 1004.241091] usb 1-1.4: new full-speed USB device number 11 using ehci-pci
[ 1004.322757] usb 1-1.4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 1.00
[ 1004.322773] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1004.322776] usb 1-1.4: Product: Crasile
[ 1004.322778] usb 1-1.4: Manufacturer: Flipper Devices Inc.
[ 1004.322780] usb 1-1.4: SerialNumber: flip_Crasile
[ 1004.323474] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```
And /dev/sdc is gone.
```
lsblk -f /dev/sdc
lsblk: /dev/sdc: not a block device
```
I thought maybe I could write a ducky script to save to the virtual disk, but I guess not?

Willy-JL, 2024-03-23:
you just discovered the whole idea that we have been working on with javascript implementation.

in javascript, you can combine badusb and mass storage (via usbdisk module), we have some working example scripts on our discord, or you can look at the examples in apps/Scripts on flipper for badusb and usbdisk, then figure out how to combine them.

closed 2024-03-23 by Willy-JL

Willy-JL, 2024-03-23:
also fixed qflipper preventing mass storage from working in 48e37ef0f7d624d348488877a50e72f18410c643

leafy-psx, 2024-03-23:
Thanks, yeah I saw the javascript example on Talking Sasquatch. For some reason I thought once you selected the disk image it stuck. Anyway, to make javascript work in Linux, you need node.js installed. I recently installed it to build Chromium but it isn't installed by default. But with ducky you can just run shell commands and that usually works on most Linux distros once a terminal is open. I will have to look at those examples, thanks.