Next-Flip/Momentum-Firmware #60: JS: Timer Module

corrupt3k, 2024-03-30:
### Description of the feature you're suggesting.

I have a script that runs indefinitely. It would be nice to easily show how long it has been running with the option to pause it, if needed.

Willy-JL, 2024-04-05:
making the js runner app pause the script doesnt seem that intuitive or feasible. some way to get the current running time for the script might make sense. as for pausing, hard to say. this idea needs to be thought out more

Chrisrdouglas, 2024-04-20:
I actually like the idea of being able to read the current time. Might be helpful to do something like checking if it's 5 PM on M-F then run a badusb script that makes my computer hibernate or something.

I'm sure other people could find uses for it

Willy-JL, 2024-08-23:
this is actually being implemented in ofw https://github.com/flipperdevices/flipperzero-firmware/pull/3841

looks very promising and well structured

Willy-JL, 2024-11-11:
available in [mntm-008](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-008)

closed 2024-11-11 by Willy-JL