Next-Flip/Momentum-Firmware #61: JS: Math library not working right.

corrupt3k, 2024-03-30:
### Describe the bug.

Generating random numbers within a defined range it not working. Numbers are generated outside of the range sometimes.

### Reproduction

let math = require("math");

function getRandomArbitrary(min, max) {
    return math.floor(math.random() * (max - min + 1)) + min;
}

while (1 === 1) {
	let randomIndex = getRandomArbitrary(0, 23);
	print(randomIndex);
	delay(1000);
}

Willy-JL, 2024-04-02:
can you check on #67 ?

Willy-JL, 2024-05-12:
Fix was released in [mntm-002](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-002)

PR #67 (2024-04-02): Explain RNG differences, Fix JS Math.random()
# What's new

- Explain differences between `furi_hal_random_get()` and `rand()`
- Add `FURI_HAL_RANDOM_MAX` define
- Fix JS `Math.random()` with `FURI_HAL_RANDOM_MAX`
- Fixes #61 

-----
# For the reviewer

- [x] I've uploaded the firmware with this patch to a device and verified its functionality
- [x] I've confirmed the bug to be fixed / feature to be stable
closed 2024-05-12 by Willy-JL