Next-Flip/Momentum-Firmware #69: Only showing default level 1 animations

DingleDangleDoff, 2024-04-03:
### Describe the bug.

With the animation pack set to default and unlock animations set to off. It only shows the level 1 default animations. 

### Reproduction

1. Turn on
2. Idk its not really something that gets reproduced, just a passive thing

Kuronons, 2024-04-03:
What is your actual level ?

Kuronons, 2024-04-03:
As of today, April 3, 2024, on Momentum fw, default manifest is set to display original Flipper animations as follows :
(if unlock animations is set to OFF, animation will only shows if its corresponding minimum level is reached)

- L1_Waves_128x50 -> Min level: 1
- L1_Cry_128x64 -> Min level: 1
- L1_Laptop_128x51-> Min level: 2
- L1_Senpai_128x64 -> Min level: 3
- L1_Sleep_128x64 -> Min level: 5
- L1_Recording_128x51 -> Min level: 7
- L1_Read_books_128x64 -> Min level: 8
- L1_Mad_fist_128x64 -> Min level: 9
- L1_Furippa1_128x64 -> Min level: 10
- L2_Wake_up_128x64 -> Min level: 12
- L1_Kaiju_128x64 -> Min level: 13
- L1_Sad_song_128x64 -> Min level: 14
- L1_Mods_128x64 -> Min level: 15
- L1_Boxing_128x64 -> Min level: 16
- L3_Freedom_2_dolphins_128x64 -> Min level: 16
- L1_My_dude_128x64 -> Min level: 17
- L1_Painting_128x64 -> Min level: 18
- L2_Furippa2_128x64 -> Min level: 20
- L2_Dj_128x64 -> Min level: 21
- L2_Coding_in_the_shell_128x64 -> Min level: 22
- L1_Leaving_sad_128x64 -> Min level: 23
- L2_Hacking_pc_128x64 -> Min level: 24
- L2_Soldering_128x64 -> Min level: 25
- L2_Secret_door_128x64 -> Min level: 26
- L3_Lab_research_128x54 -> Min level: 27
- L3_Hijack_radio_128x64 -> Min level: 28
- L3_Furippa3_128x64 -> Min level: 30

Willy-JL, 2024-04-03:
yep. most likely youre on level 1. level up to get more, or enable unlock animations. this is intended behavior. sadly, not enough animations to have one for each level, but its almost like that.

closed 2024-04-03 by Willy-JL