Next-Flip/Momentum-Firmware #74: flipper zero pocsag firmware corrupt

sysadminpower2019, 2024-04-04:
### Describe the bug.

Hello when i use the pocsag pager with the hamradio network all i get is corrupted data. See screenshot. I'm not sure what to try

### Reproduction

Try to receive a message
[image]

Willy-JL, 2024-04-04:
Did it give an api mismatch message?

sysadminpower2019, 2024-04-05:
Where would i check that?

https://ci3.googleusercontent.com/meips/ADKq_NYp6yFpb3df_Jg_S-uY9C8C6aLUXJrjZeN3cvem06u9L5qRC5R0dFHDvzaIgCVf9V_s1t0wX2ZrWqwvXH41LCuGW9x2tYVtsrwV5jeXSzflRvr-28vdt7Wi6823dIO6yE71FZ6YLHdjLBHw2ok_kb5wQS_jUqKknyK2nB7GM67-Rprq25Ezkf5Rks4wHmto7gdFzK70gTHWiDNbAhbInDYGoqV_s3pYwpDTaLsxV4Wtn2Cq7EP3jgQ=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABS6BREHS3L3ZJE7USLY3W4WBA5CNFSM6AAAAABFX6RBCCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTZPRZQG.gif>Message
ID: ***@***.***>

Willy-JL, 2024-04-05:
opening the app it would say "app too new" or "firmware too new" or something like that, showing 2 api versions.

sysadminpower2019, 2024-04-06:
I dont see anything like that. looking at the info it says version 0.1 
developed by xmasterx and shmuma but thats about it

Willy-JL, 2024-05-29:
cant replicate. the app (from lab.flipper.net, this is not an app we include in momentum) works fine on my end. maybe it is simply that your pocsag message is jumbled data and the app cannot show it correctly. or maybe it is a very old version left from some other firmware / other tiem you installed it, i remember a bug like that 1+ year ago.

also, there is no need to use the dedicated app. momentum includes the pocsag protocol inside the main subghz app. i just tested and the same messages are received by the pocsag app and momentum's subghz app.

pocsag pager app from lab.flipper.net:
[image]

subghz app in momentum (remember to configure correct modulation (usually FM95 works) and frequency!)
[image]

closed 2024-05-29 by Willy-JL