Next-Flip/Momentum-Firmware #75: storage.virtualQuit does not free the virtual reference.

jamisonderek, 2024-04-05:
### Describe the bug.

In my script: I use storage.virtualInit(image) and .virtualMount(). Write some content to "/mnt/bug.txt".  Call storage.virtualQuit.  When I call usbdisk.start(image) the app hangs; blocked on a resource since virtualQuit did not free the resource.

I think the end of `js_storage_virtual_quit` needs to do something like:
```c
    if(storage->virtual) {
        storage_file_free(storage->virtual);
        storage->virtual = NULL;
    }
    mjs_return(mjs, MJS_UNDEFINED);
``` 

### Reproduction

The following script will demonstrate that after a `storage.virtualInit` that `storage.virtualQuit` does not release resources, so `usbdisk.start` will block.

```js
let usbdisk = require("usbdisk");
let storage = require("storage");

let image = __dirpath + "/bug.img";

print("Checking for Image...");
if (storage.exists(image)) {
  print ("Storage Exists. Removing.");
  storage.remove(image);
}

print ("Creating Storage...");
usbdisk.createImage(image, 8 * 1024 * 1024);

print ("Copying Payload...")
storage.virtualInit(image);
storage.virtualMount();
storage.write("/mnt/demo.txt", "Hello");
storage.virtualQuit();

print("Attaching storage...");
usbdisk.start(image);
// BUG: The `usbdisk.start` never returns.

print("Waiting for storage to detatch...");
while (!usbdisk.wasEjected()) {
    delay(1000);
}
print("Stopping storage...");
usbdisk.stop();
print("Done.");
```

### Target

js

### Anything else?

I confirmed that adding the `storage_file_free` call in `js_storage_virtual_quit` makes the JavaScript works as expected.

Willy-JL, 2024-04-05:
great catch! will have it fixed asap

Willy-JL, 2024-04-05:
fixed via 2f06a01f646e948dd508a4a909dab5141cf3931f, thank you!

Willy-JL, 2024-04-05:
Oh my god hahahahha

I need more sleep, I'm so sorry this was my bad

Willy-JL, 2024-05-12:
Fix was released in [mntm-002](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-002)

closed 2024-05-12 by Willy-JL