Next-Flip/Momentum-Firmware #78: Adding two new .ir files to inflared folder and two new .txt remotes base on those

viralatadr, 2024-04-05:
### Describe the enhancement you're suggesting.

Hi, i just recently find and create two new .ir files for two Washlet/Bidet brands and also create two remotes for those file to put in inflared/remote. Is there a way that this could be part of Momentum? 
[image]
[image]
[image]

viralatadr, 2024-04-05:
.

Willy-JL, 2024-04-05:
we don't want to include extra files by default, we are a firmware after all. we only include example files typically, and for ir that is the samsung remote. IIRC this was part of a bounty on our discord, i suggest you share them in the script sharin channel, and maybe contribute to a file dump like uberguidoz'

closed 2024-04-05 by Willy-JL

viralatadr, 2024-04-05:
I had never done a pull request bofore. Today i did two. It was fun. Thank you for the recomendation. Now lets see if they aprove it.

viralatadr, 2024-04-05:
> we don't want to include extra files by default, we are a firmware after all. we only include example files typically, and for ir that is the samsung remote. IIRC this was part of a bounty on our [...]

Did it in the Uberguidoz and logickworkshop repo

Willy-JL, 2024-04-05:
nice!