Next-Flip/Momentum-Firmware #79: JS: storage APIs not able to read/write from "/mnt/" path

jamisonderek, 2024-04-05:
### Describe the bug.

In a JavaScript, if you use storage.read/write/copy with `/mnt/` (virtual storage) the Flipper will crash with message `Flipper crashed but has been rebooted. applications\services\storage\storage_glue.c`.  If you attach a debugger it appears the issue is `storage_file_ref` is coming back NULL?

```c
void* storage_get_storage_file_data(const File* file, StorageData* storage) {
    StorageFile* storage_file_ref = storage_get_file(file, storage);
    furi_check(storage_file_ref != NULL);
    return storage_file_ref->file_data;
}
```

### Reproduction

Calling `storage.write`, `storage.read` or `storage.copy` with a value of `/mnt/demo.txt` crashes.

```js
let usbdisk = require("usbdisk");
let storage = require("storage");

let image = __dirpath + "/bug.img";

print("Checking for Image...");
if (storage.exists(image)) {
  print ("Storage Exists. Removing.");
  storage.remove(image);
}

print ("Creating Storage...");
usbdisk.createImage(image, 1 * 1024 * 1024);

print ("Copying Payload...")
storage.virtualInit(image);
storage.virtualMount();
let d2 = storage.write("/mnt/bug.txt", "Hello");
print(d2.length);
storage.virtualQuit();
```

### Target

js

### Anything else?

I think a similar script code was working with dev build from yesterday?  I was using `storage.read("/mnt/result/output.txt");` to read a file from the mounted image.

Willy-JL, 2024-04-05:
Got another similar report from a user on discord... it's quite puzzling to be honest. I'll have a look as soon as I have time today

jamisonderek, 2024-04-05:
Thanks.  I spent some time trying to debug it but didn't see anything obvious.  It impacts both read and write, I was hoping the answer was `/mnt/` changed to something else.  :(
I confirmed the above code works fine in commit 0e9c0c2bb0b9f039df31d94934564d9ce01ecf88.

jamisonderek, 2024-04-05:
It is a regression introduced by the fix for bug #75 .  `virtualMount` is freeing the storage instead of `virtualQuit`.

closed 2024-04-05 by Willy-JL