Next-Flip/Momentum-Firmware #89: Tools > Authenticator causes reboot

drowl87, 2024-04-11:
### Describe the bug.

mntm-001 crashes/reboots every time I try to open the Authenticator app

### Reproduction

1. Switch on
2. Go to Apps
3. Go to Tools
4. Open Authenticator app

SuperJakov, 2024-04-11:
Try the dev version and provide the logs.

ANiceCupOfTea, 2024-04-11:
I don't know how to view/extract logs from the device but I ran into the same issue today and retested on the latest Dev build '9b3f7f49'. It still crashes.

Willy-JL, 2024-05-01:
Can't replicate. Might be caused by an incompatible or corrupted config file. I suggest copying your apps_data/totp folder to a pc and then deleting it, that will probably fix it.

Cc @akopachov if you want to investigate more

akopachov, 2024-05-01:
Highly-likely it is due to too old config file, that was created using crypto algo v1. The support of this algo was dropped at v5 in favor of having more memory for non-obsolete features. If it is acceptable - just drop the `totp.conf` file. If it is not acceptable - build or download artifacts suffixed with `old-crypto-support` from [Authenticator repo](https://github.com/akopachov/flipper-zero_authenticator/actions/workflows/nightly-build.yml) it will allow to migrate from crypto v1 to the latest one.

closed 2024-05-12 by Willy-JL