Next-Flip/Momentum-Firmware #94: When is the next official momentum release?

mindstormpro3, 2024-04-15:
When is the next official momentum release?

N3ggabyte, 2024-04-16:
Facts, my an does anyone know a mag spoof that works or how to truly use it?

mindstormpro3, 2024-04-16:
No go look it up

ToxxikHalo, 2024-04-16:
N3ggabyte... there's a couple boards available, rabbit labs is probably the best one... 
Magspoof is a kind of overhyped attack and not really applicable in the real world... and it's still underdeveloped code so far and it's gonna take a lot more work if it happens at all and the card security tech is already pushing beyond its capabilities for the most part... there are cards still susceptible but not many and none that are a reliable attack vector.... 
Look up some videos on YouTube there's some good ones that break it down so you get a better understanding of the complications...

N3ggabyte, 2024-04-16:
ToxxikHal😇. Much appreciated for your candor, and additional insight. I
have the rabbit labs magspoof card, finally understood how to get the
official Dev to flash. I just can’t find much on this specific specific
application. Maybe it worked better on the older cards a not the chipped
cards. Probably will have to look into it. I’ve made a mindnode of each app
category and what they do w a video example attached. Which is kinda
handy.  Has anyone been able to upload new JavaScript payloads to it? Also
I stumbled across a oxbow with 4G and LTE and thinking about adding it to a
board

Anyone else having isssues with their ocaso being zero. I watched Sasquatch
but didn’t fix anything.

Willy-JL, 2024-04-16:
@mindstormpro3 the next release is when it's ready.

closed 2024-04-16 by Willy-JL