Next-Flip/Momentum-Firmware #95: The light does not turn off.

NunoFilipePereira, 2024-04-16:
### Describe the bug.

The light does not turn off mesoq ue is stinky for this. I already updated 3 times but the problem continues. In this way the battery loses performance and it is not very attractive to always walk with the light access.

### Reproduction

.

### Target

.

### Logs

```Text
.
```

### Anything else?

.

ZProLegend007, 2024-04-16:
The backlight? Check your display backlight timeout setting numpty XD

NunoFilipePereira, 2024-04-16:
Yes . I already tried to redefenir. Change from firmware to another and Ecran erases. but with this not.

Willy-JL, 2024-04-16:
@NunoFilipePereira momentum > misc > screen > rgb backlight OFF. It explicitly asked you if you had installed the hardware mod for rgb backlight when you enabled it. You do not have the hardware modification but still enabled it. What did you think was going to happen?

closed 2024-04-16 by Willy-JL