Next-Flip/Momentum-Firmware #101: Opening subghz read settings causes unresponsiveness

ZProLegend007, 2024-04-19:
### Describe the bug.

When opening the subghz app and going to read, the config, after waiting a couple seconds the flipper becomes unresponsive. (Does not happen on read RAW config.) I am on dev firmware, commit c4c1870.

### Reproduction

Turn on flipper, open subghz through any method, go to read, then config, then wait a couple seconds.

Willy-JL, 2024-04-19:
what modulation are you using? and are you using an external module? do you have a custom subghz configuration file? if so, share it

ZProLegend007, 2024-04-19:
No customer configuration file, I think it's on HND_1 modulation atm.

ZProLegend007, 2024-04-19:
Funny you mention modulation... after quickly switching modulations the issue is gone.

ZProLegend007, 2024-04-19:
It's worth noting the there is slight noticeable lag in the config menu...

Willy-JL, 2024-04-19:
its hard to diagnose like this. try on a freshly formatted sd card, or different sd card altogether

ZProLegend007, 2024-04-19:
I tried a fresh install. It seems to only happen when the modulation is set to HND_1.

Willy-JL, 2024-07-10:
Indeed, HND_1 modulation was problematic and had too high data rate for flipper hardware and current protocols set. Removed in Unleashed and here too. Can of course be traded if you have specific needs for it by using settings user file.

Willy-JL, 2024-07-14:
Just gonna keep open until release since it's still in 004 :D

closed 2024-07-29 by Willy-JL