Next-Flip/Momentum-Firmware #106: js:usbdisk wasEjected returns always true (macOS mount)

rbartolome, 2024-04-23:
### Describe the bug.

Using usbdisk.wasEjected() in javascript returns always true when mounted in macOS. Mass Storage app works correct.

### Reproduction

1. Use Javascript code
```
let usbdisk = require("usbdisk");
let image = "/ext/apps_data/mass_storage/AImage.img";
usbdisk.start(image);
delay(5000);
print("USB Disk activated");
while (true) 
{
  print(usbdisk.wasEjected());
  delay(1000);
}
```
2. Run Code while Flipper is attached to macOS
3. The Image is mounted correct
4. The while loop always prints true even when the volume is mounted and unmounted

### Anything else?

I'm running the latest dev build

Willy-JL, 2024-04-23:
Interesting find... to be completely honest, the wasEjected logic is a bit of a hack because I couldn't find any reliable metric for when ejection happens, so I remember using the next best things from the events that the code handled, and seems like you just found a side effect

rbartolome, 2024-04-23:
Maybe it's my mistake. I did play a little bit around. If I use the code above alone it works. If I use it with badusb before it didn't. Maybe you can close this issue.

```
let badusb = require("badusb");
let usbdisk = require("usbdisk");
let dialog = require("dialog");

let imageName = "0D";
let image = "/ext/apps_data/mass_storage/" + imageName + ".img";
 
// **************************
// Remove the following Code it worked
print("init BadUSB...");
badusb.setup(
{
  vid: 0x05ac,
  pid: 0x0250,
  mfr_name: "Apple Inc.",
  prod_name: "Apple Keyboard",
  layout_path: dialog.pickFile("/ext/badusb/assets/layouts", ".kl")
});

print("Connecting BadUSB...");
while (!badusb.isConnected()) 
{
  delay(500);
}

print("deinit BadUSB...");
badusb.quit();
// Remove the Code above it worked
// **************************

print("init USB Disk...");
usbdisk.start(image);
delay(5000);
print("USB Disk activated");

while (!usbdisk.wasEjected()) 
{
  print(usbdisk.wasEjected());
  delay(1000);
}

usbdisk.stop();

print("Done");
```

rbartolome, 2024-04-23:
When I add a delay(5000); after badusb.quit() it works. Sorry for wasting your time.

Willy-JL, 2024-04-23:
Still, this deserves some more investigation. But very good that you found more info on it! I'll try to have a look when I get the chance