Next-Flip/Momentum-Firmware #107: Flipper app

ziggyiggi, 2024-04-23:
### Describe the bug.

This MAY be the wrong location, but I've recently noticed the flipper zero app store is not compatible with this version of momentum.

[image]
[image]

### Reproduction

Connect flipper to flipper zero app and go to hub to try and install apps

### Anything else?

Momentum ROCKS!

rbartolome, 2024-04-23:
I did open an issue at the official flipperdevices repo [#3606](https://github.com/flipperdevices/flipperzero-firmware/issues/3606). In general you only see apps if you are running the latest release version. The issue is closed by flipperdevices.

ziggyiggi, 2024-04-23:
Ah! Just noticed I'd accidentally installed  the dev version. Apologies!

closed 2024-04-23 by Willy-JL