Next-Flip/Momentum-Firmware #109: External Speaker support for audio

ZProLegend007, 2024-04-24:
### Describe the enhancement you're suggesting.

It would be abolutely _amazing_ to have external speaker support for apps like music player and especially WAV player. This would be so soo good and would allow for further hardware customisation. I'm assuming its possible as i've tested an external speaker with the PWM generator in the signal generator app in GPIO, I'm guessing it would have to be app specific or perhaps a firmware toggle for outputing audio to specific GPIO pins.

Would be greatly appreciated! Let me know how doable this is!

### Anything else?

I know it's said a lot but from all the firmware I've tested, you guys have nailed this. It's up to date, and super stable compared to other firmwares. Good job!

MARTINrichard05, 2024-05-01:
I d'ont remember with exact pin, put there is one that looks like wire with the buzzer : 
https://www.youtube.com/watch?v=HYNNOseehV0
i've already tested with headphones, the sound quality is not bad

ZProLegend007, 2024-05-01:
I've wired up an external aux jack from internals, works well enough. I believe in that video he has done some internal wiring to make the audio come from those pins... Thanks tho!

Willy-JL, 2024-05-01:
Wav player also outputs to a GPIO pin, check its readme

closed 2024-05-06 by ZProLegend007