Next-Flip/Momentum-Firmware #111: Flipper Crashes When Opening NFC App or using NFC Radio. [Dev version of the firmware]

Ligerbot, 2024-04-29:
### Describe the bug.

I am using the dev version of the firmware. When I open the NFC app with debug on the flipper crashes. When I use the NFC radio when debug is off the flipper also crashes. The error message it always give is "Out of memory". I would fix this on my firmware but I can not really code in C++ and also I have no idea how flipper apps work. I am also not sure if it is because my NFC radio isn't attached properly because I am in the middle of opening the flipper up. It is almost unscrewed so the radio might not be making electrical contact, but I don't think that would cause an out of memory error.

### Reproduction

1. Open NFC
2. Use any NFC function that uses the radio.

Ligerbot, 2024-04-29:
Update : It is not my radio, because I tried holding it together and it still was crashing.

Willy-JL, 2024-04-29:
Settings > system > heap trace = off

Ligerbot, 2024-04-30:
There is no off setting but I set it to main and it worked. Thanks a lot.

closed 2024-04-30 by Ligerbot

Willy-JL, 2024-04-30:
It should be on None @Ligerbot . Does it crash on none?