Next-Flip/Momentum-Firmware #112: HIgh battery consumption

gchokov, 2024-05-02:
### Describe the bug.

Flipper is using about 10% battery a day when idle. I used to run Xtreme and didn't see this behavior. 
What I did in the last three days: 
- Read bunch of EM4100
- Wrote some EM4100
nothing else...

Can't provide mode details than this.

### Reproduction

Flipper is using about 10% battery a day when idle. I used to run Xtreme and didn't see this behavior. 
What I did in the last three days: 
- Read bunch of EM4100
- Wrote some EM4100
nothing else...

Can't provide mode details than this.

Willy-JL, 2024-05-02:
Settings > power > battery info, how much consumption at idle with screen backlight off?

Also do you usually keep a module plugged into gpio?

gchokov, 2024-05-02:
Thanks for the response, @Willy-JL. Your work is highly appreciated. 
I've left the Flipper for 5 minutes untouched, just on the battery screen. Consumption is at 24mA. I think it should be sub 10mA?

Willy-JL, 2024-05-02:
Was the screen light off for that 24mA?

gchokov, 2024-05-02:
Yes, no screenlight. It dropped to about 16mA now, after 10 minutes of no use. Probably something running somehow in the background.

Willy-JL, 2024-05-02:
Perhaps. Settings > system, check:

Debug off

Heap trace none

Sleep method default

Log level none

After this power off the flipper and leave it off for 20 seconds, then power on again

gchokov, 2024-05-02:
Well, it looks like it was the Debug being ON. I've had DEBUG to ON due to reading some RAW NFC Data. 
After setting the DEBUG to OFF and rebooting, the consumption is to "Napping ~3mA"). 

Thanks @Willy-JL .. just got you some Ko-Fi.

closed 2024-05-02 by gchokov

Willy-JL, 2024-05-02:
Yep debug will disable deep sleep CPU mode. Deep sleep means debugger cannot be attached, so enabling debug will stop deep sleep so debugger can be attached, something along those lines. Glad it's fixed :D and thanks!