Next-Flip/Momentum-Firmware #124: Electra protocol support in v2

devpwnz, 2024-05-16:
### Describe the bug.

I am trying to read an Electra tag, but it does not identify it at all. With older firmware, it would show up as EM4100. If I remember correctly, they would coincide, but the Electra tags have some extra bytes to the serial. Am I doing something wrong?

### Reproduction

Open RFID - Scan Electra Tag - Nothing Happens.

Leptopt1los, 2024-05-16:
@Willy-JL @devpwnz need debug logs first

Willy-JL, 2024-05-16:
@devpwnz 

Rfid > read raw, post the dump here

Connect to https://lab.flipper.net, go to CLI, type "log debug", try to scan the tag on flipper, then post the log output here

devpwnz, 2024-05-17:
I'm attaching the RAW readings here.
[Electra-raw.zip](https://github.com/Next-Flip/Momentum-Firmware/files/15344934/Electra-raw.zip)

And the log.
[debug-electra.txt](https://github.com/Next-Flip/Momentum-Firmware/files/15345011/debug-electra.txt)

@Leptopt1los @Willy-JL

devpwnz, 2024-05-17:
I have an addendum. At the beginning, when I got the Flipper, I tried scanning the Electra tag, which was read as a an EM4100. I then tried to write it back to the tag, which probably worked and would explain the last 0's from my tag :)

Leptopt1los, 2024-05-17:
@devpwnz is it original electra tag? definitely not copy? can you attach pic of it?

devpwnz, 2024-05-17:
I have an addendum. At the beginning, when I got the Flipper, I tried scanning the Electra tag, which was read as a an EM4100. I then tried to write it back to the tag, which probably worked and would explain the last 0's from my tag :)

@Leptopt1los

Leptopt1los, 2024-05-17:
@devpwnz this is extremely unlikely. as far as I know, electra tags do not implement the t5577 protocol

as for your tag, it's quite easy to correct its reading. the problem is that this solution will be less elegant (you will have to store an additional 5 bytes). epilogue filler does not actually affect the detection of the tag by the intercom, so theoretically we can transmit anything instead. but I am disgusted by the idea that a copy taken by a flipper will differ from the original tag, even if it does not affect its performance

it would be nice to find a second tag that won't be readable either, so that we have a more relevant sample to analyze the need to adjust the solution. Could you purchase a second copy of the key for the same intercom in the electra office? I think the data from reading the second instance can help us to say more confidently whether this situation is the norm or the exception

as for the immediate solution, I suggest you install the firmware version 01, read the tag, remember its id. then flash back to 02 and manually add the electra tag with id XX XX XX XX XX 7E 1E FF, where XX is the em4100 id, read on version 01

by the way, information about the success/failure of emulating this tag on your intercom would also be quite useful

Willy-JL, 2024-05-17:
Thank you for looking into it and for the very detailed explanation Lept ❤️🙏

devpwnz, 2024-05-18:
@Leptopt1los 

I have read another tag yesterday, that was enrolled at the same time as my tag and on the same device and it was read correctly. 

If know that Electra also sells tag cloners, it-s called PRG.400 and uses PRO-TAG software. They don't sell 2 types of tags, so I'm guessing that their tags are also writable. We have a place where we copy keys and whatnot, and they used to clone Electra tags on other Electra tags, and after a little while, they clone it on .. clones, not Electra brand.

Thus, I'm inclined to think that my read/write actually worked, but it left zeroes at the end.

I see you are proposing to leave 7E 1E FF at the end, but when presenting the tag in debug mode, they appear at the beginning. Also, my tag still works on the original reader, as it is.

Thank you very much for your response, and @Willy-JL for being so interested in getting to the bottom of this.