Next-Flip/Momentum-Firmware #127: Adding an item to the Main menu

ifuchs, 2024-05-17:
### Describe the enhancement you're suggesting.

I have a playlist that requires 11 key presses to access and run (in SubGHZ app). I was wondering if there was any way to simplify access by adding it to the Main menu or to have another "Frequently accessed"  menu item that had these items under it.

Willy-JL, 2024-05-17:
A "playlist" doesn't use the "SubGHz" app. Do you mean a file that requires the "SubGHz Playlist" app? Or a saved file that runs in the "SubGHz" app?

Regardless, you can add files to your favorites (from desktop, press down, press right to go to SubGHz tab, find the file, hold ok on it, Favorite it). Or you can set it as a Desktop Keybind (Settings > Desktop > Setup Keybinds).

Willy-JL, 2024-05-17:
Those are the alternatives, but to answer your question: currently we have no plans to add file support to the main menu, except for JavaScript files which we are considering adding support for.

ifuchs, 2024-05-17:
Sorry if I was less than clear. From the Main menu, I go to Apps->SubGhz->SubGhz Playlist->myPlaylist. Is there a way to shorten this?

Willy-JL, 2024-05-17:
Yes, you can use one of the 2 ways I described

Willy-JL, 2024-05-17:
Or as a third option you can add the subghz playlist app itself to main menu, in Momentum > interface > main menu > add app > external app > subghz > subghz playlist

ifuchs, 2024-05-17:
OK, got the 3rd option which adds the Sub-Ghz app to the main menu. I wanted to get the playlist there though this a an improvement for sure. 
I tried one of the other alternatives (#2) unsuccessfully. I did Settings->Desktop->Keybinds Setup-> Keybind type->Press->Keybind Key->Up->External App->Sub-Ghz->SubGhz Playlist->my playlist and I am taken back to Keybinds Setup. I obviously went wrong somewhere along the line.

As to alternative #1, I am not clear on how to do what you described to get my playlist on the main menu.

Willy-JL, 2024-05-17:
When you go to add an app to the main menu, you are given the choice of main app or external app. If you choose external app, you can select the subghz playlist app itself and have it on the main menu.

Also, the Keybind was set correctly, you need to exit the app and go back to the desktop to active the Keybind you configured. Looks like now, when pressing the up button on the desktop, it will open your subghz playlist. Also, keybinds let you choose between Open File and External App, the difference being that External App sil Popen SubGhz playlist (the app), while Open File will directly open your myplaylist file

ifuchs, 2024-05-17:
I am able to add the SubGhz app to the main menu (per your alternative #3). I tried #2 again to no avail. When you refer to the "desktop", is that the same as the main menu? Also, per alt #2, if I simply go to the file and hold on it there is no option to create a "favorite" so I assume I am in the wrong menu to do that.

Willy-JL, 2024-05-17:
Desktop is the one with the dolphin animations and the sd card icon in the top left and battery in the top right

ifuchs, 2024-05-17:
Ahh. OK, I set my playlist as a favorite. Is that somehow accessible from the main menu?

RogueMaster, 2024-05-17:
> Ahh. OK, I set my playlist as a favorite. Is that somehow accessible from the main menu?

From desktop, you press down

ifuchs, 2024-05-17:
Fantastic! Thanks for your patience.

closed 2024-05-18 by Willy-JL