Next-Flip/Momentum-Firmware #128: All sub ghz frequencies blocked

Messha777, 2024-05-18:
### Describe the bug.

Since update all subghz frequencies are blocke, lost region provisioning

### Reproduction

Switch on

Willy-JL, 2024-05-18:
Need to update with qflipper or web updater to provision the region file. Or you can bypass region in momentum > protocols > subghz bypass region

Willy-JL, 2024-05-18:
It's almost as if the error screen tells you exactly what you need to do to fix the issue, nifty isn't it?

closed 2024-05-18 by Willy-JL

Messha777, 2024-05-20:
I was able to fix this by region bypass. Now whenever I read a raw signal
the spl number doesn't move up and it's not saving the signal or saving
such a small timeframe that it can't be sent.

Willy-JL, 2024-05-21:
you need to lower your threshold value, also theres probably no signal being transmitted so no samples being collected

Willy-JL, 2024-11-19:
it says "Reinstall firmware with Web/App". qflipper is a flipper companion app. reinstalling with qflipper fixes the issue too. the point is, region file is missing, reinstalling firmware with webupdater / qfli8pper / mobile app will fix it. because these methods will deploy the region data file.