Next-Flip/Momentum-Firmware #129: Nightstand Clock: Alarm Clock Function?

MrBreadWater, 2024-05-19:
### Description of the feature you're suggesting.

We have a nightstand clock app, but no alarm? I feel like an alarm feature would be trivial to implement and a great little upgrade to the nightstand clock. 

### Anything else?

Honestly it wouldnt even need to run in the background or anything, just go off at the set time if you have have the clock open.

StellarStoic, 2024-08-17:
I've been waiting for this for more than a year already and it boggles my mind why this hasn't been added by the official flipper team. 

There's a countdown app which triggers an alarm. 
We have 2 clock apps and none of them have an alarm 🤦🏻

Willy-JL, 2024-08-23:
@kakrzewiaczek in #184:

> In the application, you can add alarms, which you can set a name for.
>
> => We set when the alarm should go off (time and day/days of the week) and confirm.
>
> Even if Flipper Zero is locked, the alarm will ring (this can be turned off in Momentum settings). Additionally, in the application, you can turn vibrations on and off and set the alarm sound.

Willy-JL, 2024-09-18:
https://github.com/flipperdevices/flipperzero-firmware/pull/3906

stefanodelbosco, 2024-11-01:
> [flipperdevices/flipperzero-firmware#3906](https://github.com/flipperdevices/flipperzero-firmware/pull/3906)

Merged 2 day ago

Willy-JL, 2024-11-02:
yeah, and im merging it in #287 , already works great

Willy-JL, 2024-11-03:
now merged into dev branch, can use it in dev build. will be in 008 shortly

Willy-JL, 2024-11-11:
available in [mntm-008](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-008)

closed 2024-11-11 by Willy-JL