Next-Flip/Momentum-Firmware #131: "universal" remote control

leonardpitzu, 2024-05-20:
### Description of the feature you're suggesting.

I have 2 remotes for 2 different road barriers, 2 different door tags and 4 different access card, all different standards. It would be nice to have a "remote control" app on the Flipper Zero and link each button to a different remote, tag, card or whatever, WITHOUT switching between the different apps/protocols. It's just a usability thing as the functionality behind would be the same but would allow to have a "one remote GUI to rule them all". Think of switching the remotes while driving if an explanation is needed.

Willy-JL, 2024-05-20:
There's also more specific apps like cross remote that links remote in a chain, xremote that maps the flipper buttons to ir signals, subghz remote does the same for subghz files, there's subghz playlist and nfc playlist... and worse comes to worst, there's the favorites menu... lots of great ways to automate, I'm sure you'll find one that suits you )

leonardpitzu, 2024-05-20:
quac! - this is it. Thanks. Issue solved!

closed 2024-05-20 by Willy-JL