Next-Flip/Momentum-Firmware #133: Marauder

1c0wb0y1, 2024-05-21:
### Describe the bug.

The marauder companion seems to not be capturing pcaps properly. It collects them but they all say 0 bytes. It could be just me but I’ve seen it on other FW so I didn’t know if there was an issue or if it’s just my end. 

### Reproduction

Open marauder. Scan ap. Select ap. Attack pmkid. Collect pcaps. Analyze pcaps.

Willy-JL, 2024-05-21:
Apps > gpio > esp > esp flasher > quick flash > your board type > marauder

CuriousCocainist, 2024-06-15:
Hi, 
Sorry to squeeze in, but have a similar issue.
1) Marauder flashed to dev board the way you mentioned (went through, although I got "E (217) esp_core_dump_flash: No core dump partition found!")
2) During the first use set to save pcaps (and logs) to flipper sdcard
3) Following attack ("EAPOL:" displayed few times on f0) no pcap files saved (no files of any type in "pcaps" folder), despite logs were saved successfully in "logs" folder

Willy-JL, 2024-06-15:
@CuriousCocainist you enabled saving on flipper side, but the board doesn't have it. Usually it's on by default from what I know but looks like it's off for you. Marauder > settings: save pcaps> run command

CuriousCocainist, 2024-06-16:
Scratch that, I'm an idiot... :')
Got totally confused by the other ways you can access files on f0... Using "the browser" (down arrow from the main screen) allows me to see captured pcaps, while through marauder app ("View log from" or "Load Evil Portal HTML file") they look empty. 
Thanks for your help @Willy-JL  :)

Willy-JL, 2024-07-10:
If you still have issues, please follow this @1c0wb0y1:
- Apps > GPIO > ESP > ESP Flasher > Quick Flash > Your board type > Marauder (has evil portal)
  - If you see "failed to mount sd card" this is NORMAL
  - Once the numbers stop counting down, you will see some text that doesn't seem to make sense, that's probably the marauder ascii art, means it finished successfully
- Apps > GPIO > ESP > Marauder
  - At the bottom, Save to flipper sd card > PCAPS Yes > Logs your choice
  - Settings: SavePCAP > Save

For now I am closing this issue due to inactivity, feel free to send more comments if this is not sufficient to resolve your issue.

closed 2024-07-10 by Willy-JL