Next-Flip/Momentum-Firmware #134: Apps (hub) in companion app incompatible firmware

Figure-itout, 2024-05-21:
### Describe the bug.

When you try to install/update apps with the mobile companion app you get a message to update the firmware.
momentum 001 works to install (not updating), momentum 003 give the error.
[image]

### Reproduction

@Flipper
1. Switch on bluetooth

@Companion app
1. Connect to flipper
2. go to tab Hub
3. go to Apps

### Target

mntm-003

### Anything else?

Updating apps also didn't work with mntm-001.
Tried stock firmware 0.101.2, works fine, even updating apps.
Mobile phone is Motorola Edge 30 neo, Android 14.

Drotnip, 2024-05-21:
I've been having the same issue started 2 days ago. I factory wiped and reinstalled off works. Custom fw does not.

Willy-JL, 2024-05-21:
The api version changed. It works in dev build.

Willy-JL, 2024-05-29:
update to mntm-004

closed 2024-05-29 by Willy-JL