Next-Flip/Momentum-Firmware #135: "MPU Fault, possibly stack overflow" after trying to start Evil portal Server

vgiamp10, 2024-05-22:
### Describe the bug.

when i click "Start portal" on the Evil portal app the flipper crashes and throws up the error "MPU Fault, possibly stack overflow"
i used the ESP Flasher app on the flipper zero and with the marauder app it works so the only app not working with the devboard is the evil portal one
im using the official wifi devboard from the official flipper zero shop
[image]

### Reproduction

honestly, i don't know, from what i saw online im the only one having this problem but ill try
1. New install of momentum
2. flash the marauder firmware with the esp flasher app
3. reboot the flipper zero
4. open the evil portal app
5. start evil portal
6. ???
7. CRASH!

Willy-JL, 2024-05-26:
are you using apps > gpio > esp > evil portal ?

Genmutant, 2024-05-26:
I get the same error every time using DEV 7FC115A and "apps > gpio > esp > evil portal".

vgiamp10, 2024-05-26:
> are you using apps > gpio > esp > evil portal ?

yes i am

nicolastoews, 2024-06-06:
I have the same issue :(

gh0stinth3sh3ll, 2024-07-05:
I too am having this issue. Can anyone please help??

JohnHeberty, 2024-08-05:
Set AP Name and select your html before starting the portal, it worked for me.

vgiamp10, 2024-11-15:
> Set AP Name and select your html before starting the portal, it worked for me.

THANKS IT WORKED

closed 2024-11-15 by vgiamp10