Next-Flip/Momentum-Firmware #138: SubGHz file geolocation data change in Momentum FW

bip3r, 2024-05-26:
### Describe the bug.

Grabbed and decoded a SubGHz signal.
Edited it in TXT and added a geolocation data to it: 

e.g.:
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok270Async
Latitute: xx.312873
Longitude: yy.561488
Protocol: KeeLoq
Bit: 64
Key: AA BB CC DD EE FF 00 11
Manufacture: NICE_Smilo

After saving the file and then viewing the "Geographic info" in SubGHz app of Flipper it shows:
"Captured at: xx.312874, yy.561487" 

So it adds +1 to lattitude and substracts -1 from longitude, however the file data stays unchanged (..73, ..88) - checked it twice.
A curious one, however doesn't affect much, as the difference of  xx.zzzzz(z+/-1) doesn't change the real location drastically :)

### Reproduction

Grab some SubGHz signal, decode.
Edit the file on laptop/desktop, add geoloaction data.
Run SubGHz app in Flipper
Choose the grabbed file
Choose Geographic Info

### Target

SubGHz geolocation data change

Willy-JL, 2024-05-27:
my guess is float precision error. you manually inserted a value that a float is not capable of representing in ram

bip3r, 2024-05-28:
RGR that. Thx for replay. Not a real issue as the shift is just 4-5 meters in real life :)

closed 2024-05-28 by Willy-JL