Next-Flip/Momentum-Firmware #141: Customize the infrared remote button's positions on the screen using tables

Questwalker, 2024-06-04:
### Describe the enhancement you're suggesting.

Currently, custom infrared remotes only display their button in a long list on the screen. This is a unattractive way to sort the buttons, so I propose that some optional settings `row`, `column`, `rowspan`, and `columnspan` can be added to each button that will allow users to customize the positions of the buttons on the screen using a table-like system.

These settings should only be optional, in order to support backward compatibility.

### For example:
```
Filetype: IR signals file
Version: 1
# 
name: POWER
# 
name: Vol+
column: 0
columnspan: 1.5
# 
name: Vol-
column: 1.5
columnspan: 1.5
# 
name: SOURCE
column: 0
columnspan: 2
# 
name: MT
column: 2
# 
name: OK
column: 0
row: 3
rowspan: 2
# 
name: RE
row: 3
column: 2
# 
```
Would create something similar to this:

[image]

### Anything else?

I have created this to mimic the ways other graphics tools create tables. I understand if this kind of approach is not possible, but I would at least like some way to compact and arrange the buttons on the remote interface in order to make the experience easier to use.

Willy-JL, 2024-06-04:
There's quite a few apps that make infrared signals easier to use, check out Xremote, Quac!, CrossRemote, IR Remote. In regards to this feature request in particular, I think it's too much complexity for the added benefit it brings, considering those apps I mentioned already provide good alternative approaches. It's a no from me, but feel free to ask upstream if you're not satisfied with the alternatives

closed 2024-06-04 by Willy-JL

Willy-JL, 2024-06-04:
Quac is in apps > tools, also please update to version 004