Next-Flip/Momentum-Firmware #147: JS: keyboard.text only displays and prompts the first time.

jamisonderek, 2024-06-14:
### Describe the bug.

If you call keyboard.text twice, it no longer works.  Older builds (like mntm-004) would display both `Question` and `Answer`.  Now I get prompted for Question (the first use of keyboard.text) but I don't get prompted for Answer.

mntm-dev [14-06-2024] / 906a1e7a [65.0].

### Reproduction

Create the following keyb.js file, copy it to SD card, and then run it.

let keyboard = require("keyboard");
keyboard.setHeader("Question");
let question = keyboard.text(100, "", true);
print(question);
keyboard.setHeader("Answer");
let answer = keyboard.text(100, "", true);
print(answer);

### Target

js

Willy-JL, 2024-06-14:
hmmm this is most likely due to recent event loop changes, it refactored inner view dispatcher workings... from a first guess im assuming event loop is receiving the stop event from previous view dispatcher in the new view dispatcher due to being on the same thread. it is a odd usecase, but this sould not be the case. ill investigate further and upstream. thanks for the report

Willy-JL, 2024-06-14:
fix submitted here https://github.com/flipperdevices/flipperzero-firmware/pull/3711

Willy-JL, 2024-06-14:
fixed via eaf1a5dd3d27dda4dd5809d0cf281050a2a53362

closed 2024-06-14 by Willy-JL