Next-Flip/Momentum-Firmware #151: Flipper Zero battery low screen disappears

I-Flip-Zeros, 2024-06-29:
### Describe the bug.

When Flipper Zero hits 0% battery charge, a screen pops up asking the user if they want to cancel the shutdown, acknowledge the shutdown, or shutdown now. However, the screen disappears, not allowing the user to make a choice. The Flipper Zero then defaults to shutting down in 90s.

### Reproduction

1. Flipper Zero hits 0% battery charge. 
2. Screen appears that asks the user what they want to do
3. Screen disappears after 1-2s 
4. User cannot make a choice, and Flipper Zero turns off in 90s.

https://github.com/Next-Flip/Momentum-Firmware/assets/124535943/596da1f9-f64f-428e-933f-e1f7e5b15de9

I-Flip-Zeros, 2024-06-30:
Forgot to mention, the bug only happens when Flipper Zero is restarted

I-Flip-Zeros, 2024-07-06:
What additional info is needed?

Willy-JL, 2024-07-06:
Label means that it's not clear yet what is happening or how to fix it. Either give any info you can think of, or we will eventually look into it with the info you already provided

I-Flip-Zeros, 2024-07-07:
The battery low screen disappears when it isn't supposed to. I can't make a choice regarding the shutdown, because the screen disappears too soon. The screen isn't supposed to disappear, as seen in the above video

Willy-JL, 2024-08-23:
@I-Flip-Zeros can you please try on latest dev build? many things changed related to power service

I-Flip-Zeros, 2024-08-23:
Yeah, updating now. I'll drain the battery and let you know if the bug is fixed.

I-Flip-Zeros, 2024-08-23:
Issue is resolved on dev 55fd64ee

I-Flip-Zeros, 2024-08-23:
👍

closed 2024-08-27 by Willy-JL