Next-Flip/Momentum-Firmware #166: NFC Crash

s1rleblanc, 2024-07-18:
### Describe the bug.

Im trying to clone a NFC Hotel Card and every time i try to do that all the flipper crash

[image]

here is a screenshot.. im tried to find the path, but i can't.

### Reproduction

1. Go to NFC
2. Read
3. Put card in flipper back
4. Crash

Willy-JL, 2024-07-18:
This seems like a software shortcoming of desfire protocol implementation. Please try to replicate on official firmware, and if the issue persists consider reporting the issue to https://github.com/flipperdevices/flipperzero-firmware including some debug logs (can view from cli on lab.flipper.net)

s1rleblanc, 2024-07-18:
Thank you i will try meanwhile i past the cli debug log of momentum:

148328 [34m[D][NfcScanner] [0mFound 6 children
148372 [34m[D][Nfc] [0mFWT Timeout
148374 [34m[D][Nfc] [0mFWT Timeout
148407 [34m[D][Nfc] [0mFWT Timeout
148439 [34m[D][Iso14443_4aPoller] [0mRead ATS success
148473 [34m[D][Iso14443_4aPoller] [0mRead ATS success
148520 [34m[D][Iso14443_4aPoller] [0mRead ATS success
148523 [34m[D][EMVPoller] [0mSend select PPSE
148529 [31m[E][EMVPoller] [0mFailed to parse application
148553 [32m[I][NfcScanner] [0mDetected 1 protocols
148671 [33m[W][ViewPort] [0mViewPort lockup: see applications/services/gui/view_port.c:303
148690 [34m[D][Iso14443_4aPoller] [0mRead ATS success
148698 [34m[D][MfDesfirePoller] [0mRead version success
148702 [34m[D][MfDesfirePoller] [0mRead free memory success
148706 [34m[D][MfDesfirePoller] [0mRead master key settings success
148710 [34m[D][MfDesfirePoller] [0mRead master key version success
148715 [34m[D][MfDesfirePoller] [0mRead application ids success
148718 [34m[D][MfDesfirePoller] [0mSelecting app 0
148722 [34m[D][MfDesfirePoller] [0mReading app 0
148734 [34m[D][MfDesfirePoller] [0mCan't read file 0 data without authentication
148737 [34m[D][MfDesfirePoller] [0mSelecting app 1
148742 [34m[D][MfDesfirePoller] [0mReading app 1
148758 [34m[D][MfDesfirePoller] [0mCan't read file 1 data without authentication
148761 [34m[D][MfDesfirePoller] [0mSelecting app 2
148766 [34m[D][MfDesfirePoller] [0mReading app 2

hope help or work for something.. im will try OFW.

Willy-JL, 2024-10-30:
@s1rleblanc can you try latest dev build? Some changes have been made to the 14a poller

bmt626, 2024-11-05:
I just flashed mntm-dev-2e7eedf2 and a Desfire card still causes a crash when trying to read it

EDmitry, 2024-11-05:
Hitting the same crash when trying to save the card. When emulated it works fine, the problem is only when saving.