Next-Flip/Momentum-Firmware #169: FW comparison to Xtreme

sizzlik, 2024-07-25:
Hi,
in the past i was using the Dev version of the Xtreme FW, sadly the main head clara is going through some RL problems and it seems the project is on hold for now the last 3 month..unsure if or when it comes back or if there are even any plans.
So i started to look around for other compareable FW and stumbled over this one.
Can anyone tell me how similar or not similar the two are? Some major things that one can do, but the other doesnt? Or stuff one can do better than the other? Im not looking for a specific use case, just overall swiss army knife features/apps and an active dev team.
Thanks for any help or opinion.

SavvaGor1207, 2024-07-25:
Here is a great README: [here](https://github.com/djsime1/awesome-flipperzero#firmwares--tweaks

Additionally the repo is a great collection of Flipper resources.

sizzlik, 2024-07-25:
ah..i had that repo starred already..should have taken alook there. Thanks for the reminder. -closed-

closed 2024-07-25 by sizzlik

Willy-JL, 2024-07-25:
@sizzlik if you look at the commit activity of Xtreme, it was mostly me developing and maintaining it. We had disagreements in Xtreme between me as developer and Clara as project owner, so she removed me. Since I was the lead developer for over a year and really enjoyed developing firmware, I continued my work under this new Momentum project but maintaining the same codebase (just removing NSFW content which was due to her involvement). For the most part, Momentum is what Xtreme would still be if I was never removed, since I would have continued developing with mostly the same philosophy and goals over there if it hadn't ended like that. But also the community and atmosphere around Momentum seems markedly better and there's a lot more cool people from the community interested and contributing here than there ever was in Xtreme.

ZProLegend007, 2024-07-25:
Momentum is a direct continuation of Xtreme. I switched a while ago when it stopped updating and this one is basically the same thing renamed and better. Best firmware I can find personally 😁

brneor, 2024-07-25:
> NSFW content

As a new Flipper owner and new Momentum user I was just wondering what kind of NSFW content is even possible on the Flipper and why would one even want to do that... 🤔

Willy-JL, 2024-07-25:
@brneor NSFW was actually one of the core selling points when Xtreme just started in late 2022 / early 2023, it featured a 30 level system with one new NSFW animation for each levelup, aswell as many system icons and onscreen text replaced with lewd versions. I got my flipper in early 2023 and found out about Xtreme from a Talking Sasquach video, but what caught my eye was that he showed one code diff for the level system, which had refactored a 30 else-if chain to a for loop, which is of course much better code quality, so I thought I would find some capable people there. I didn't know at the time, but now I know that actually the 30 else-if chain came from RogueMaster firmware and it was refactored in Xtreme (so it wasn't fixing some awful code from stock firmware, I now know those folks write great code), and most of the other optimization claims from early XFW development were just removing buggy and excessive code that came from RogueMaster and replacing it with the originals from Unleashed or Official firmware. Nevertheless I joined Xtreme shortly after they had implemented a way to disable the NSFW content, which was done in a simplistic manner which I later reworked into what we know today as Asset Packs (and the NSFW assets that Xtreme had at the beginning became a separate asset pack). As for who would want NSFW on flipper, I mean you know how teenagers are, and we all know that lots of flipper users are young people that saw a master hacker device on tiktok, you get the gist xD

brneor, 2024-07-25:
@Willy-JL well, ty for your work in Momentum! Can I get in touch to ask some questions on how to get involved in firmware development and how to contribute in general?

Willy-JL, 2024-07-25:
@brneor The discord is always open. For contributing, best way is via PRs. For getting started, I suggest reading the official flipper firmware documentation, looking at existing code, and following Derek Jamison videos on YouTube.