Next-Flip/Momentum-Firmware #171: invalid manifest

Shadow-Blitz, 2024-07-29:
### Describe the bug.

Hello,

I am having an issue upgrading from mntm 004 to mntm 005. The update.fuf file gives me an invalid manifest error. I have downloaded the update using the .zip option.

The md5 of the update.fuf file is.
91e71f9c7946b79e578d54c69c0d7df1

The md5 of the zip folder containing the update is.
8808fd629fe9eb459dd6425f5cf75caa

I do not have access to the desktop qflipper app as all I have is my Android phone and tablet.

I have been utilizing the USB mass storage ability on the flipper zero to transfer the extracted zip file from my phone to the flipper zero via a USB cable. And then mounting the mass storage image and copying the folder into the update folder and updating the flipper using the update.fuf file. This has been working fine ever-since the first mntm version (mntm-001)

Please let me know if you need any further information.

### Reproduction

Prerequisite:
Have a mass storage image created using "USB mass storage" on the flipper zero

Reproduction

1. On Android device download the update zip file.
2. Extract the zip file.
3. Connect Android device to flipper zero using a USB C to C cable.
4. On flipper zero go back to initial USB mass storage app and select the image to be mounted as a USB storage device.
5. Copy the update folder to the flipper zero mass storage.
6. Unmount the mass storage device on Android
7. Disconnect flipper zero from Android device
8. On flipper zero mount the storage medium by going to, Browser apps_data/mass_storage/file.img
9. Long hold on centre button on flipper zero, select "Mount"
10. Long hold centre button (again)  select "Actions" select "Copy"
11. Copy the folder into the update folder by pressing right after selecting "Copy" and entering the "update" folder
12. Long hold the centre button and select "Actions" and then "Paste"
13. Enter into the newly pasted update folder
14. Select update.If
15. You will get invalid manifest error

Shadow-Blitz, 2024-07-30:
The reason why I don't use the official flipper app (if you are talking about the one on Android) is because the Bluetooth transfer speed is extremely slow. After half an hour the transfer was only at 25%.

Unless there is a way to update it using Android and a USB cable connectig the 2 devices. Please let me know. As what I am doing right now is the only way to update it at a sufficient speed.

Willy-JL, 2024-08-23:
@Shadow-Blitz is this still an issue? is it possible that your android is messing with the `resources.tgz` file? have seen other occureences where .tgz files were extracted or renamed and caused confusion

also, please try with latest dev build

WillyJL, 2025-07-20:
closing as stale

closed 2025-07-20 by WillyJL