Next-Flip/Momentum-Firmware #173: qFlipper crashes on macOS when attempting to install

linkslice, 2024-07-30:
### Describe the bug.

I've updated to the latest version of qFlipper, flashed from extreme to latest stock firmware with no change. Was even able to flash back to xtreme. I've tried multiple versions of momentum with no change (005, 004, 003, etc..)

### Reproduction

1. download latest momentum firmware
2. click install from file
3. select the 005.tgz file
4. *poof* qFlipper is gone

### Logs

```Text
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               qFlipper [9772]
Path:                  /Applications/qFlipper.app/Contents/MacOS/qFlipper
Identifier:            com.yourcompany.qFlipper
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-07-30 09:59:03.2542 -0500
OS Version:            macOS 14.6 (23G80)
Report Version:        12
Anonymous UUID:        C80A2282-EAF3-C058-8C1D-753F535B7640

Time Awake Since Boot: 53000 seconds

System Integrity Protection: enabled

Crashed Thread:        13  Thread (pooled)

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000038
Exception Codes:       0x0000000000000001, 0x0000000000000038

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [9772]

VM Region Info: 0x38 is not in any region.  Bytes before following region: 4334075848
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10254c000-1045e8000    [ 32.6M] r-x/r-x SM=COW  /Applications/qFlipper.app/Contents/MacOS/qFlipper

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19271ab28 kevent_id + 8
1   libdispatch.dylib             	       0x1925cc2ec _dispatch_kq_poll + 228
2   libdispatch.dylib             	       0x1925cb830 _dispatch_event_loop_poke + 340
3   QuartzCore                    	       0x19abd3bb8 CA::Context::commit_transaction(CA::Transaction*, double, double*) + 16592
4   QuartzCore                    	       0x19aa2d874 CA::Transaction::commit() + 648
5   QuartzCore                    	       0x19ac03454 CA::Transaction::flush_as_runloop_observer(bool) + 136
6   CoreFoundation                	       0x19283887c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
7   CoreFoundation                	       0x192838768 __CFRunLoopDoObservers + 536
8   CoreFoundation                	       0x192837480 CFRunLoopRunSpecific + 684
9   HIToolbox                     	       0x19cfe119c RunCurrentEventLoopInMode + 292
10  HIToolbox                     	       0x19cfe0e2c ReceiveNextEventCommon + 220
11  HIToolbox                     	       0x19cfe0d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
12  AppKit                        	       0x196096cc8 _DPSNextEvent + 660
13  AppKit                        	       0x19688d4d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
14  AppKit                        	       0x196089ffc -[NSApplication run] + 476
15  qFlipper                      	       0x102663c6c QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 1792
16  qFlipper                      	       0x103aad798 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 540
17  qFlipper                      	       0x103aa523c QCoreApplication::exec() + 128
18  qFlipper                      	       0x10255fbc4 main + 372
[... 430 lines cut ...]
Graphics: Apple M2 Pro, Apple M2 Pro, Built-In
Display: ASUS VN247, 1920 x 1080 (1080p FHD - Full High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Apr  4 2024 20:57:11 version 23.30.58.0.41.51.138 FWID 01-baea9d27
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB 10/100/1000 LAN
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 Hub
USB Device: Flipper Ashqu0s
USB Device: HD Pro Webcam C920
USB Device: Keychron K1
USB Device: G403 HERO Gaming Mouse
USB Device: EXTERNAL_USB
USB Device: USB2.0 Hub
USB Device: Apple Watch Magnetic Charging Cable
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

Willy-JL, 2024-07-30:
Hm looks like a null pointer dereference. Try reinstalling qflipper and redoenloading the tgz file

N-coder82, 2024-07-30:
Sometimes these crash reports can include PII, I don't think this one does, but a warning for the future

linkslice, 2024-07-30:
I did re-install qFlipper, and tried updating to latest version before reporting. It only happens with momentum. xtreme, and stock firmwares install ok. I've also unisntalled/re-isntalled flipper, but it still crashes when attempting to load momentum.

Willy-JL, 2024-07-31:
Did you try downloading the momentum tgz again?

linkslice, 2024-07-31:
I did, I just tried again with firefox and it worked. Safari was doing something awry, it seems.

closed 2024-07-31 by linkslice

Willy-JL, 2024-07-31:
Glad to hear you got it working!

jeffreyjackson, 2025-12-27:
Coming here to confirm that Safari is indeed doing something weird on the download.  I tried with a different browser and install started without crashing qflipper immediately.

WillyJL, 2025-12-28:
macOS likes to decompress files after downloading. Zip files usually get extracted automatically, which is unbelievably stupid. Firmware downloads are tgz, which is a tar archive with gzip compression. What Safari is doing is decompressing the gzip stream but not unpacking the tar archive, all while leaving it with a tgz extension. So it's tar data in a tgz filename, and when qflipper goes to treat it as a tgz it crashes.
Safari is an utterly unserious browser and should never be used for anything.

corygehr, 2026-09-17:
If you're stubborn and must use Safari (or have no other choice, for some reason), you can turn that weirdness off under Safari > Settings > General > Uncheck "Open 'safe' files after downloading."

WillyJL, 2026-09-17:
> If you're stubborn and must use Safari (or have no other choice, for some reason), you can turn that weirdness off under Safari > Settings > General > Uncheck "Open 'safe' files after downloading."

does it really solve the problem raised here? from the name of the setting, it seems like it would fix the first issue (files like .zip being extracted automatically), rather than the second and main focus of this report (.tgz files being decompressed to .tar data in-transport but still saved as .tgz, its not "opening" that happens after the download is finished, safari is wrongly treating the gzip compression of the original file as transport compression, it thinks the original file on the server was uncompressed)

corygehr, 2026-09-17:
> does it really solve the problem raised here? from the name of the setting, it seems like it would fix the first issue (files like .zip being extracted automatically), rather than the second and [...]

Surprisingly, yes.

I agree you wouldn't expect that setting name to dictate behavior for this specific situation, but I didn't have the crash described in this Issue after unchecking the box, re-downloading, and trying again. I figured I'd comment with my resolution since this is the first thing which popped up when I was looking for answers.