Next-Flip/Momentum-Firmware #181: storage.mkdir() doesn't work on virtualMount

khimaros, 2024-08-13:
### Describe the bug.

take for example this JS script:

```
let usbdisk = require("usbdisk");
let storage = require("storage");
let image = "/ext/example.img";
usbdisk.createImage(image, 32 * 1024 * 1024);
storage.virtualInit(image);
storage.virtualMount();
storage.mkdir("/mnt/payload/");
storage.copy("/ext/example.txt", "/mnt/payload/example.txt");
```

the `storage.copy()` fails because `/mnt/payload/` doesn't exist.

### Reproduction

1. add the script above to a file in scripts
2. launch the script

Willy-JL, 2024-08-13:
Trailing slashes are not supported on flipper. It's mkdir /mnt/payload

khimaros, 2024-08-13:
thank you, that was the issue.

a bit off topic, but do you know if there is a way to `storage.remove()` recursively or otherwise to `storage.copy()` in a way that merges into an existing directory?

Willy-JL, 2024-08-13:
not at this time. expanding the features of storage module is on the todo list but not high in priority

closed 2024-08-18 by khimaros