Next-Flip/Momentum-Firmware #185: Furi_check failed when doing MIFARE Fuzzing with MIFARE DESFire

Clats97, 2024-08-19:
### Describe the bug.

When using the MIFARE Fuzzer, and selecting a DESFire saved capture, upon selecting either test values or random values or uploading from a file, the Flipper crashes and says "furi_check failed" after rebooting. Running most recent version of firmware. This doesn't happen with classic or ultralight 

### Reproduction

1. Turn on flipper
2. Select "MIFARE Fuzzer"
3. Select a DESFire capture
4. Select "test values" or "random values" or "upload from file"
5. Press attack
6. Flipper restarts and says "Furi_check failed" 
[image]

Willy-JL, 2024-08-19:
i believe this app is only for mifare classic, but even then crashing when presented wrong card is not ideal

Clats97, 2024-08-19:
I figured it could be used for other types of cards because of the "load from file" option but I'm probably mistaken

[signatureImage]